wled/WLED #1860: How to setup custom IR codes

xkolm036, 2021-04-04:
HI,
I wanted to setup my custom IR codes. So i define new codes in ir_codes.h and edit decodeIRCustom in ir.cpp like this:
`bool decodeIRCustom(uint32_t code)
{
  switch (code)
  {
  //just examples, feel free to modify or remove
  case IRCUSTOM_ONOFF:
    toggleOnOff();
    Serial.println("on/off");
    break;
  case IRCUSTOM_BRIGHTER:
    bri = 255;
    Serial.println("bri=255");
    break;
  case IRCUSTOM_DARKER:
    bri = 0;
    Serial.println("bri=0");
    break;

  default:
    return false;
  }
  Serial.println("returned true");
  lastValidCode=code;
  return true;
}
`
But now when button on my remote controller is pressed I can see (in serial monitor) that case for specifying button is called but nothing happened (brightness is still on the same level in all three cases ). Can anyone help me how to setup custom IR correctly ?

Thank you.

Whole ir_codes.h and ir.cpp files: 
[ir.zip](https://github.com/Aircoookie/WLED/files/6255400/ir.zip)

My WLED verion: 0.12.0
My device: nodeMCUv2

Aircoookie, 2021-04-08:
Hi!
Please make sure you call `colorUpdated(NOTIFIER_CALL_MODE_BUTTON);` everytime you change the `bri` (or color and fx) value :) That should fix it.

xkolm036, 2021-04-09:
Hello,

When I call colorUpdated(NOTIFIER_CALL_MODE_BUTTON); It works perfectly :)

Thank you for help me.

closed 2021-04-09 by xkolm036