wled/WLED #1861: Lights are not saved after they are discovered

myrulezzz, 2021-04-04:
**Describe the bug**
After installing app on iphone my lights are discoverable and working.However after i close the app and open it again lights are not saved in my list and i have to rediscover them

**To Reproduce**
Steps to reproduce the behavior, if consistently possible

**Expected behavior**
A clear and concise description of what you expected to happen.

**WLED version**
 - Board: [e.g. Wemos D1, ESP32 dev]
 - Version [e.g. 0.10.0, dev200603]
 - Format [e.g. Binary, self-compiled]

**Additional context**
Anything else you'd like to say about the problem?

Thank you for your help!

myrulezzz, 2021-04-05:
Hi after restarting my device the issue is reolved.
Thanks

Aircoookie, 2021-04-07:
Strange, normally the device should save the list once you exit the app. Glad that it's working now though!

closed 2021-04-07 by Aircoookie