wled/WLED #1865: Usermod interface into runtime configurable pin definitions

huggy-d1, 2021-04-05:
When working with usermod's prior to v0.12.x, pin definitions naturally fell to the usermod code with other changes as required for the usermod.

With v0.13.x? and above, it would be fantastic to have usermod pin assignments also runtime configurable through some usermod allowances that have also enabled UI, API, and wifi interactions.

It might be super simple, and it might be really hard.  I am sorry I cannot tell which it is at the moment.

Aircoookie, 2021-04-06:
Thank you for the suggestion!

Luckily, it is very possible! In fact, blazoncek has already added the capability for 0.12, but I patched it out for the time being because I wanted to release 0.12 quickly and not add a new capability for the usermod "API" that I likely would need to change for the new 0.13 logic.

huggy-d1, 2021-04-06:
That is fantastic news which I expect will make many a usermod creator and usermod user very happy.  It will enable us to try multiple I/O pin configurations without recompiling and keeping up with which version is currently compiled and installed/loaded.

Thank you and @blazoncek for excellent efforts in runtime pin configuration, be it built-in, or usermod(s).

Will 0.13 allow multiple usermods installed together with ease?  Or, does an earlier version already enable that and I missed it?

Aircoookie, 2021-04-06:
Awesome!
Multiple usermods are already quite easily possible with v2 usermods (introduced with 0.11 I believe). Either copying in the usermod and adding it into `usermods_list.cpp` works, or the usermod creator can make it even easier and add a compile time define. In that case, you just need to define e.g. `-D USERMOD_DALLASTEMPERATURE` as an extra build flag in your PIO environment and the mod is compiled in :)

huggy-d1, 2021-04-06:
What is this _usermod_creator_ of which you reply in all of your awesomeness!???!!

Aircoookie, 2021-04-06:
Could be you 😉

huggy-d1, 2021-04-06:
OHHHH, I thought you meant there was this "tool" called "usermod creator" that when invoked just needs -D USERMOD_USERMODENAME and magically the PIO override ini file generates the usermod enabled WLED installable .bin.  Now I see you mean the usermod author / developer / coder / maker / hacker .. that kind of creator.

In my head, I had this idea that VS Code / Platform IO worked there magic by executing a script named "usermod creator" where it started automatically inserting usermod's into the usermods_list.cpp if they were provided on the command line, and inserting the usermod code into WLED for building, and it somehow enabled the API to route commands to the right usermod, and UI pages automatically created and bonded with the proper usermod.

That's what I thought you were saying was already available.

And if it is, well, that's simply awesome and please tell me how and I'll try to update the wiki to reflect it.

blazoncek, 2021-04-07:
I am already working on it (usermod config page). :)

blazoncek, 2021-04-13:
First general alpha versions available for testing from my fork (dev branch).
Currently 3 usermods have been converted to be compatible: Temperature, 4 Line display and Auto Save.

blazoncek, 2021-05-16:
This is now merged into master and will be available in 0.13.

closed 2021-06-29 by Aircoookie