wled/WLED #1866: Wled flashing in Segment 5

brmeadows2841, 2021-04-06:
I have an ESP32s wroom  setup using the following GPIOs pins:

 GPIO16 (using 50 ws2811) 
IO17 (using 100 ws 2811)
I05 ( using 1 dumb led set as PWM white) 
IO18 ( using 50 ws2811) these four segments work as expected

When I add addition segment 4 ( ws2811 50 leds) on io19, 21 22, or 23 I get a flash in the effects even on solid.
The flash on segment 4 happens even if I turn off segment 0,1,2,3, as stated above I have tried segment 4 on the pins 19,21,22,23 of course I changed the pin assignment in the LED settings page to match the segments.

**To Reproduce**
Add segments beyond four segments.

**Expected behavior**

I expected the solid effect on segment four to be solid without a flashing.

**WLED version**
 ESP32s wroom

downloaded binary from WLED

 - Format [e.g. Binary, self-compiled]

**Additional context**
Anything else you'd like to say about the problem?

Thank you for your help!

brmeadows2841, 2021-04-06:
> Hi!
> I suspect you might have overlapping segments, which can lead to such flashing. Can you send some screenshots of your Segments in the main UI (each segment's start + stop LED) as well as the "Hardware Setup" section of your LED settings?

Hi!
I suspect you might have overlapping segments, which can lead to such flashing. Can you send some screenshots of your Segments in the main UI (each segment's start + stop LED) as well as the "Hardware Setup" section of your LED settings?

—

 

Thanks for you attention to this issue

 

 

I have included the API command 

 

 

 

{"on":true,"bri":128,"transition":7,"mainseg":0,"seg":[{"id":0,"start":0,"stop":50,"grp":1,"spc":0,"on":true,"bri":83,"col":[[255,111,0,0],[0,0,0,0],[0,0,0,0]],"fx":2,"sx":255,"ix":104,"pal":0,"sel":false,"rev":false,"mi":false},{"id":1,"start":50,"stop":150,"grp":1,"spc":0,"on":true,"bri":255,"col":[[66,69,153,0],[0,0,0,0],[0,0,0,0]],"fx":57,"sx":128,"ix":128,"pal":0,"sel":false,"rev":false,"mi":false},{"id":2,"start":150,"stop":151,"grp":1,"spc":0,"on":true,"bri":255,"col":[[204,41,16,0],[0,0,0,0],[0,0,0,0]],"fx":100,"sx":128,"ix":128,"pal":0,"sel":false,"rev":false,"mi":false},{"id":3,"start":151,"stop":201,"grp":1,"spc":0,"on":true,"bri":255,"col":[[204,27,0,0],[0,0,0,0],[0,0,0,0]],"fx":90,"sx":128,"ix":128,"pal":0,"sel":false,"rev":false,"mi":false},{"id":4,"start":201,"stop":251,"grp":1,"spc":0,"on":true,"bri":255,"col":[[94,153,0,0],[0,0,0,0],[0,0,0,0]],"fx":0,"sx":128,"ix":128,"pal":0,"sel":true,"rev":false,"mi":false},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0}]}

Aircoookie, 2021-04-06:
I have checked and there are luckily no overlapping segments in your setup. How are your LED strips configured in LED settings?

brmeadows2841, 2021-04-06:
Is this what you are asking about?

Aircoookie, 2021-04-06:
Sorry, but I cannot seem to see your attachment. Please try uploading it directly to github instead of e-mail reply, hopefully it will work then :)

brmeadows2841, 2021-04-06:
@

brmeadows2841, 2021-04-06:
Sorry I’m new to the GitHub  app

Aircoookie, 2021-04-06:
No worries at all!
Your LED settings are correct too. So just to confirm, the strip that is flashing is No. 5, which is set to pin 22 in the screenshot?

Two things you could try:
- Use pin 26. That one is proven to work well, although it should technically not matter.
- Swap the pins of the flashing strip and one of the good ones. If the flashing one is ok now and the previously good one flashing, it is a WLED / pin issue. Otherwise, it is likely an issue with the LEDs themselves (running WS2811 directly off an ESP32 is technically out of spec, as they require 5v data signal and the ESP only outputs 3.3v, you would likely need a level shifter to fix it). For running four digital outputs, you could consider picking up the QuinLED Dig-Quad, which has all that built in and should be back in stock soon.

brmeadows2841, 2021-04-06:
I swapped segment 0 to pin 22 and segment 4 to 16. The problem stayed with the same set of with the leds. 

I then changed them with a new set, now everything is fine. I assumed a bad set of LEDs.  Interestingly enough, I added a new set of leds to segment4, then saved and restarted WLED. Everything began to work correctly. I then added the assumed bad LED set to the end of segment 4 changed the number from 50 to 100 in hardware and the segment section. Now all 100 on segment 4 are following commands correctly  from Wled. I am thinking a bad data connection was the problem even thought I had move that connection many times trying to fine the problem.

Thanks for your software and support.

I did notice a minor quirk, when the number of LEDs are changed on a pin in hardware segment 0 get changed to the total number of LEDs.

Thanks for your guidance, I make a bad assumption

closed 2021-04-07 by brmeadows2841