wled/WLED #1867: Swap "Peek" and "Nodes" buttons?

smartroad, 2021-04-06:
**Is your feature request related to a problem? Please describe.**
In V12 there is now the nodes button when it detects other installs on the network, but seems only visible when there is a wide enough. It isn't visible on my phone screen.

**Describe the solution you'd like**
Option to swap the Peek button with the Nodes button. (I picked on the Peek as it is the least useful button for me ;) that or Timer!)

Another option would be to make the button bar being scrollable?

**Describe alternatives you've considered**
It can be accessed under the Info page but that adds extra step for such a convenient function! I have 2 installs at the moment and soon will have another couple, so a single press to bring up the list from others would be super useful!

Aircoookie, 2021-04-06:
Hi and thank you for the suggestion!
I picked the Nodes button as the one to hide when there is not enough space, as mobile users are more likely to use the app, which provides the same functionality as the instance list.
There is a little "WLED" (or other name, if you renamed the unit) text in the bottom right corner of the UI, in fact you can click that too to bring up the instance list. Might be useful.

I will think about an option to re-order or show/hide certain buttons in the top bar. In general, I am struggling to find space in the UI for new features, so maybe something like that would be helpful to de-clutter by hiding functionality you don't need.

blazoncek, 2021-04-13:
I vote for removal of peek. Might be useful for testing but has little value otherwise.

sansillusion, 2021-04-13:
I love to have peek. I dont need nodes Button onobile since I use app and
HA.

blazoncek, 2021-04-13:
A million people, a million preferences.

smartroad, 2021-04-13:
Don't know if it is possible but maybe the buttons could be customised? That way people can have the buttons they find most useful?

huggy-d1, 2021-04-30:
Maybe checkbox in WLED UI setup to set button order for enabled buttons, followed by [save & restart] so WLED can put the settings in the file store and then start using them on reboot.

Aircoookie, 2022-10-21:
I'm liking the status quo. "Nodes" is accessible one-click via the name in the bottom right corner, otherwise in the Info panel.

closed 2022-10-21 by Aircoookie