wled/WLED #1868: Top nav bar is flashing since v0.12.0 update

WoodyLetsCode, 2021-04-06:
Hello,
since the v0.12.0 update the top nav bar is flashing every time I reload the page.
This only happens in Chrome and in File Mode.

**To Reproduce**
1. Open Chrome
2. Load index.htm locally

**Expected behavior**
No flashing like in previous versions

**WLED version**
 - v0.12.0, Build 2104020

Here you can clearly see the flashing effect:
[image]

While you can see the popup you can see the top bar permanently in white:
[image]

It's only a small bug, but this flashing didn't happen in v0.11.1, so that is why I made this report. 😄 
Thank you for this cool project! 😎

Aircoookie, 2021-04-06:
Noticed that too. I will take a look next time I work on the UI 😄

softhack007, 2023-10-02:
Not observed any more in 0.14.

closed 2023-10-02 by softhack007