wled/WLED #1869: New to WLED and some questions

diggidre, 2021-04-06:
Hey guys,

I'm running ESP8266 with WS2812b and WLED 0.11.1.
It's confusing me a bit that color presets only show up on choosing some effect preset. Is this correct? In my understanding if I choose color preset it should show up immediately even if solid mode is active.
Except that controlling the LED with App is very cool feature and I love it :)
But I really need to clear up some things.
There is the streaming option in WLED (don't know exact how to call it).
I have a RPI with Hyperion running that sends over the picture data to WLED... ok nothing new :)
At the beginning I had to struggle with the connection. I played a bit with the settings and couldn't figure out how to tell WLED to listen to Hyperion. Then it was like magic :) WLED suddenly was in "Live Mode". Thought I have to enter some IP and Port in the settings of WLED. But seems WLED has a auto discover feature for that? Can someone please explain how WLED connected my Hyperion Server?
And finally I'll have to figure out the Time&Macro function. I think it needs more testing for me...
I have to mention for shutting LEDs down when tv is off I installed a master slave outlet strip. So ESP and LED switch together with the tv. But I want it more detailed and more on the software side.
For now I set 2 active time slots
1.  6h0m Preset 1 > From my understanding this means switch to preset 1 after 6am?
For preset 1 I turned LED Brightness to zero.
2. 19h0m Preset 2 > Preset 2 is some effect that has to be played when tv is turned on after 7pm.

Between 6am and 7pm if tv is turned on the leds should stay just black so when someone watches TV in the afternoon the LED stay off. After 7pm the preset 2 should be active. Hmmm...
On top there is this behaviour when the Hyperion Server is active and WLED receives data from  it. It overrites the WLED preset and shows live picture data. With that star I can turn off and on live mode but how can I automate it?
On the Pi with Hyperion I let cron execute 2 scripts on given time to start or stop hyperion with systemctl.
Problem is when the script for starting hyperion service is executed while WLED is off completly means tv is off as well I get an error message from hyperion status because WLED host is unreachable. This is normal but turning on the tv after the script on the pi has been executed results in WLED doesn't recognize hyperion anymore and starts simply with the preset I choosed to be set after boot.
 Why is the time slot not recognized or am I missunderstanding something for Time&Macro section :D
 How can I force start and stop WLED to recieve data from hyperion?
 As said this needs more testing but if someone could help me to deal with that I would be appreciated.

Aircoookie, 2021-04-06:
Hey!
Not sure if I can answer all of your points, but here is a start:
- What do you mean by "color preset"? The 3 main colors or the list of palettes underneath? Anyways, Solid mode always just displays the first color. If you want to display a palette, but not moving, use the Palette effect and set the speed slider to 0.
- Hyperion uses the 19446 port which is received by default in WLED settings. You would likely have needed to enter the WLED IP in Hyperion settings, not the other way around.
- Timed macros unfortunately don't work that way. If you set up the macro to switch to preset 1 at 6am, it will only do so at exactly 6am, if WLED was on before. If you turn on the TV (and thus, controller) at some point afterwards, the macro is not executed. For your use case, a kind of "if booting between 6am and 7pm" logic would be required, which is not currently supported. Your best bet is likely to use some kind of automation (perhaps node red running on the same server as Hyperion) to send `{"ps":1}` to the `[WLED-IP]/json` at boot (or e.g. once every minute between 6am and 7pm)
- WLED should start receiving the Hyperion data immediately after connecting to the WiFi. Maybe Hyperion stops the stream if it detects that the WLED device is unreachable, not sure on that.

diggidre, 2021-04-07:
Thx for your reply @Aircoookie .
I mean the presets underneath. I take a look into that further.

Yes thats correct. As LED Hardware I set hyperion config to WLED "protocol". Ok 19446 is the default listening port in that case.

The hyperion pi is running all the time.  I don't let the master slave outlet strip handle the rpi because I don't want to deal with eventually sdcard damage. Another advantage could be controlling WLED from a script if possible.
Maybe if I could send a command to WLED from within the script on the hyperion pi to force WLED to switch from "streaming mode" to normal mode with certain preset this would be cool.
I already searched for a command list to control WLED from another machines terminal. Would be nice to have one and play around a bit :)
For my "advanced standby" problem I configured a preset with no brightness that loads after boot. Everytime if tv is switched on the LEDs stay just black. At specific time a script on the hyperion machine starts the service and WLED will switch to streaming mode immediately.
There is just a problem with LEDs shining bright and flickering after hyperion service restart but that's a hyperion forum question I think. Must have to do something with the usb hdmi grabber. Restarting hyperion again can solve the problem. 

BTW can we have a native Knight Rider preset for WLED like hyperion has one. That would be awesome :D

closed 2021-08-22 by stale[bot]