wled/WLED #1870: Flexible Segmentierung/Gruppierung

Generalf69, 2021-04-07:
thank you so much for providing such a great tool to control tons of LEDs.

I have built my own hexagon shaped "Nanoleafs" and I have put in a 180 LED strip 2811.
Now the problem I am having with the current segement/group feature is, that it assumes all LEDs are in a row.
However, mine are not :-)
Would it be possible to have something to group specifc LEDs into a group
e.g.
groupname     address of LED
LEDgroup1 = {0, 1, 2, 3, 4, 5, 6,  56,  57, 58, 165, 166, 167, 168, 169, 170, 171, 172}
LEDgroup2 = {7, 8, 9, 10, 11, 12, 13, 14, 15, 135, 136, 137, 138, 173, 174, 176, 177, 178}
..
...

Cheers
Ralf

Generalf69, 2021-04-08:
awesome, this is exactly what I was looking for :-)  looking forward on 0.12.1 
thx again,

softhack007, 2023-01-06:
A availeable in latest releases.

closed 2023-01-06 by softhack007