wled/WLED #1876: Smooth transition between effects

TohtoriR, 2021-04-09:
I'm using effects in normal conditions for example decorative pieces or lamps with the gradient effect. This will create a nice lively effect for to letter (R) lamp i have for example. I use this to notify also. So when there is an action in the house that requires my attention the lamp goes from gradient effect to chase random effect. 

The issue is that when the new effect takes place it does not know the state of the old effect. In this case when going from the gradient to any of the sliding rainbow effects such as chase random the color will first change instantly and then the effect starts. When HA in my case command the effect back to Gradient in 30 sec its also instant and just flashes to new color instantly.

It would be nice to have some kind of a smooth transient between these effects. Either a gradual change or even better that the effect would "run out" and the base color would be fully exposed. Then the new effect would start from that by the same base color if it is some of these rainbow like effects that cycle thru different color.

What an excellent project you have here. Thanks for all your effort.
BR
-Toni

Legsmaniac, 2021-04-09:
Smith Transitions is already a thing. Find it in settings. You can adjust the timing too.

Aircoookie, 2021-04-10:
Brightness and Color transitions are supported, but not yet transitions between different effect modes.
It is already on the roadmap though :) https://github.com/Aircoookie/WLED/issues/826

closed 2021-04-10 by Aircoookie