wled/WLED #1877: Need help converting effect to WLED

nftarts, 2021-04-09:
**Take a look at the wiki and FAQ, perhaps your question is already answered!**
[FAQ](https://github.com/Aircoookie/WLED/wiki/FAQ)

**Please consider asking your question on the WLED forum or Discord**
[Forum](https://wled.discourse.group/)
[Discord](https://discord.gg/KuqP7NE)
[What to post where?](https://github.com/Aircoookie/WLED/issues/658)

**If you do not like to use these platforms, delete this template and ask away!**
Please keep in mind though that the issue section is generally not the preferred place for general questions.

fishbone-git, 2021-04-09:
https://github.com/Aircoookie/WLED/wiki/Add-own-functionality for a very loose guide.  You can basically copy a lot of what's already in FX.h and FX.cpp

nftarts, 2021-04-09:
Yes read through that its converting the code to the formatting is what im having real trouble with

nftarts, 2021-04-09:
or do i just copy and paste the contents of the ino file and use open and closing formatting?

nftarts, 2021-04-10:
willing to fund for assistance

nftarts, 2021-04-11:
can anyone assist?

Legsmaniac, 2021-04-11:
Why not start by showing us the code to the custom effect you wish to add?  😊

nftarts, 2021-04-12:
its an artists proprietary code.. I can share with dev that help

Aircoookie, 2021-04-12:
If it's not open source, I can't add it since then it would be open source.

nftarts, 2021-04-12:
Aircookie I sent code over to you via email. So you are saying you cant help in converting to wled code but im not able to release as open source?

Legsmaniac, 2021-04-12:
@nftarts Whose code is it?  Aircookie will need permission from the author of the original code if it's not open source.

nftarts, 2021-04-12:
thats not an issue code belongs to an art collective im a part of, should i send email to aircookie so he can verify?

Legsmaniac, 2021-04-12:
@nftarts Only Aircookie can answer that I guess.  I'm not a part of the WLED team myself, just an interested WLED user. (For interested, read Nosey. LOL)  What I DO know is that Aircookie is a busy chap and I'm not sure he'd have the time to chase up any verification of usage rights so I would suggest you make it as easy as possible for him and the team.  Perhaps you could link to the code somewhere?  Or a website belonging to your art collective?  I'd imagine they would need to know that everyone who is a part of this art collective who contributed to this code is on board with it being shared so if you had a website for example and the code was published on there as free for anyone to download and use, Open Source even, would make it a lot easier.  Or you could look into publishing it as Open Source on Git and then maybe submit a Pull Request?
Or as I said in my first post, just post the code on here for the team to look at.  If that's a problem then it means you're not willing to publicly share it and that would be a problem because if you want it considered to be an inclusion in WLED, the very fact that WLED is Open Source and anyone can copy, share and distribute the code means that whatever you may be trying to guard WILL get shared eventually if it was included within WLED.
Hope you see what I'm saying.  Only trying to help.  😉

nftarts, 2021-04-12:
I can verify all info on who the code belongs to but this is proprietary code used by the artist in nearly all of the sculptures. Aircookie can you please speak to this? I have sent you the code via email and can verify the collective and artist this code was written by

nftarts, 2021-04-12:
@Aircoookie i Forwarded the code again with link to artist site

Aircoookie, 2021-04-12:
Hi,
I have taken a look at your code, and it seems as it is written with a particular installation in mind and not necessarily suitable across a wide array of different LED counts. Therefore it would likely not make sense to add it as a WLED effect in the first place.

I understand that you would like me to modify this code for you to work with WLED, but only as a custom job/commision, not for the entire community, but unfortunately I do not provide that sort of thing.

If I receive e.g. FastLED code that looks amazing, is open source and works on many installations, I will more than happily adapt it for use as a WLED effect, but I do neither proprietary or specialized code. Sorry about that.

Aircoookie, 2021-04-12:
One little tip I can give you as a side note: Try optimizing the effect for memory 😉 The `cmap` array alone uses 16.8kB of RAM, which is **huge** for an LED effect. As a start, some variables would likely fit in a byte instead of int.

nftarts, 2021-04-14:
If anyone is interested in porting this for me I can fund.

closed 2021-04-20 by Aircoookie