wled/WLED #1878: Access point show up when using a simple firmware but not WLED

kelvin200, 2021-04-10:
Hi everyone, I'm new with all this and have a little issue with access point. The photos of my board are attached:

[image]
[image]

- I tried different firmwares from the latest release https://github.com/Aircoookie/WLED/releases/download/v0.12.0/WLED_0.12.0_ESP8266.bin to the custom built via PlatformIO
- I tried different power sources from USB port on the computer (5V-0.5A) to the wall charger (5V-2A)
- I tried different `default_envs` settings like `nodemcuv2` and `d1_mini`
- None of them made the access point "WLED-AP" appear and the serial console always shows the Wifi state as 0
- The LED strip connected to the board produces warm orange lights as mentioned in https://github.com/Aircoookie/WLED/wiki

[image]
[image]

I then tried a new simple project just to start an access point and test if my board is broken, and it works, even with the laptop USB port as the power source.

[image]
[image]

This is the code of the simple project

```cpp
#include <Arduino.h>
#include <ESP8266WiFi.h>

const char *ssid = "MYAP";
const char *password = "ap12345678";

int LEDPin = 13;

void setup()
{
  pinMode(LEDPin, OUTPUT);
  digitalWrite(LEDPin, LOW);

  delay(1000);
  Serial.begin(115200);
  WiFi.softAP(ssid, password); //begin WiFi access point
}

void loop()
{
  delay(500);
  Serial.print("My board WiFi Status: ");
  Serial.print(WiFi.status());
  Serial.print(", IP: ");
  Serial.println(WiFi.softAPIP());
}
```

Would anyone please instruct me on how to fix this issue? Thanks.

Aircoookie, 2021-04-10:
Hi! Sorry about the trouble, perhaps some residual flash contents have configured WLED in a way where it does not open the access point.
Have you attempted to erase the flash completely prior to flashing the binary? That often helps.
Another thing you could try is bridging the D3 and GND pins for >8 seconds. That is _supposed_ to open the access point as well.

kelvin200, 2021-04-10:
Hello, thanks for responding so quickly.

Yes, I erase the flash before flashing the binary. I do it by running `pio run --target erase` (if I use Upload function on PlatformIO) and adding `--erase-all` flag when flashing via the terminal `sudo python3 /usr/local/bin/esptool.py --port /dev/ttyUSB0 write_flash -fm dout --erase-all 0x00000 Projects/WLED/.pio/build/nodemcuv2/firmware.bin`. The issue is not fixed.

Btw, at first I did flash the board using flash_mode `qio`, which didn't even light it up. I wonder if it matters, perhaps it broke something on the board?

I just tried connecting D3 and GND for 8 seconds, the access point is still nowhere to be seen.

kelvin200, 2021-04-11:
Hi guys, I found a surprisingly simple solution for this after trying everything I could. It just needs a short delay after `handleConnection` in the main loop and it is working properly now. I guess my board takes longer to initialise a connection :man_shrugging: 

I'm not sure if it's worth investigating this little issue but these are what I found, hopefully it helps:
- I tested it again by using the latest code on master branch and the board couldn't make a connection. Then I added that single line of delay and uploaded it to the board and it worked like a charm.
- I haven't tested what would happen if reducing the delay duration to lower like a few milliseconds.
- I tried adding a delay in `initAP` function, right after the line of `WiFi.softAP(apSSID, apPass, apChannel, apHide);` but it didn't work.
- Both access point mode and wifi mode (with fixed ssid and password) don't work if the delay is not added in the loop function.
- Perhaps it only happens with my specific NodeMCU board.

[image]

Now I have a chance to see how it work, this is a wonderful project and have tons of config to play with the LED strip. I look forward to exploring more and perhaps writing my own effects. Keep up the good work, guys. Cheers!

closed 2021-04-11 by kelvin200

TUISTERa, 2021-11-17:
I can confirm that this is working, is there a way for this fix to be added to master ?