wled/WLED #1879: Feature Request - Support for variable group sizes

HammerCA, 2021-04-10:
I'm looking to create a sign with multiple letters and then inside each letter would have a variable number of LEDs.  I see I create virtual LEDs with the grouping feature but that is limited to same number of LEDs.  The feature request is to be able to support grouping of variable sizes.  This is to enable to use the effects to treat each letter of the sign as a single LED.

Aircoookie, 2021-04-10:
Hi,
that is a great idea, but would not be easy to implement.

There is a "hacky workaround" though, which relies on pixel mapping (https://github.com/Aircoookie/WLED/wiki/Mapping).
- First of all, find out what the letter with the largest amount of LEDs is. For this example, let's suppose you wanted the letters `ID` and the `I` had 5 LEDs and the `D` 8 LEDs. 
- Now, set the Grouping to that LED count, in our example 8 LEDs.
- Set your LED count in settings to that maximum per-letter number (8 LEDs) _times_ the number of letters you have. It doesn't matter that this will be more LEDs than you physically have attached. In my example, this is 16.
- Now, create a mapping file `ledmap.json` using the `http://[WLED-IP]/edit` page (see wiki article above). There you can map the LEDs each letter has to their positions and map the extra LEDs (LEDs 5-8 the grouping feature thinks belong to the `I`) and puts them to the end of the strip, behind the last letter.

Example mapping for `ID`:
```json
{"map":[0,1,2,3,4,13,14,15,5,6,7,8,9,10,11,12]}
```

Sorry that this is a bit complicated, but it should yield the desired result. If you get stuck, just send me the LED count per letter you have and I can create the ledmap file for you :)

HammerCA, 2021-04-10:
Thank you.  I will looking into the mapping functionality.  I have not tried it yet.

ripnet, 2023-06-30:
Hello. I would also like to see this implemented if possible. 

Thanks!

bipsendk, 2023-08-12:
Looks like something I could use as well.....

My challenge (simplified) - I have 15 LEDs in total. The should be "grouped" to appear as 3 groups of 5 LEDs each,
It should then work in such a way, that if I tell "virtual" LED 0 to turn on, it will actually tual on LEDs 0-4 ... If I want to turn on "virtual" LED 1 - LEDs 5-9 should turn on ...
Can/should this be done via mapping - if yes, how ?
Would it be something like 

{"map":[0,0,0,0,0,1,1,1,1,1,2,2,2,2,2]}

Now, the defined LED segment will start at offset 0, and have a length of 15. Will any effect automatically look at the LED map and figure out, that only 3 "virtual" LEDs should be included in the effect ?

blazoncek, 2023-08-12:
You don't need variable groupping (as it may not be implemented anytime soon) for that.
Just set up regular segments and use ledmaps to logically _rearrange_ LEDs.