wled/WLED #1880: Import name of realtimeMode to display the mode on LCD or TFT

AB-DMX, 2021-04-10:
Hi !! I need to know how import name of realtimeMode .... I want to display on LCD...... I make usermod and it work but just need this for finalize the project. The only thing it print is the number of realtimeMode.

Thank you !!! When is finish I upload my usermod :)

**Take a look at the wiki and FAQ, perhaps your question is already answered!**
[FAQ](https://github.com/Aircoookie/WLED/wiki/FAQ)

**Please consider asking your question on the WLED forum or Discord**
[Forum](https://wled.discourse.group/)
[Discord](https://discord.gg/KuqP7NE)
[What to post where?](https://github.com/Aircoookie/WLED/issues/658)

**If you do not like to use these platforms, delete this template and ask away!**
Please keep in mind though that the issue section is generally not the preferred place for general questions.

Aircoookie, 2021-04-10:
Hi! You could copy the switch block from [here](https://github.com/Aircoookie/WLED/blob/0e99c948d68948ec3ea43482c4b40a497c3ee552/wled00/json.cpp#L441) and just change `root["lm"]` to the string you want to set :)

AB-DMX, 2021-04-10:
Thank you Aircoookie !!!
I try this tonight :)

closed 2021-08-19 by Aircoookie