wled/WLED #1881: Sunrise and sunset with on in between

tgrantjr, 2021-04-11:
I'm looking to use this for my snake enclosures. I want to have the strip in the tank do a sunrise in the morning and stay on, then a sunset at night and they stay off till morning. Is there a way to achieve this?

blazoncek, 2021-04-13:
Curently only available in my fork (precompiled binary available on @srg74 repo) but should soon be available in main WLED.
You will need to build from source though until next release.

After you have a sunrise/sunset capable version create two triggers (one for sunrise and another for sunset) using a separate playlist for each. Look at Wiki page for JSON API regarding playlists.

tgrantjr, 2021-04-14:
I know this may sound stupid, but how do i compile this?  I have compiled in linux many times.  I dont see a bin file to flash over in the d1 mini flash program.  any way you could help me out with that if possible?  Thanks

blazoncek, 2021-04-14:
If you want to compile yourself search for Quindor on YouTube (https://youtu.be/fRxyp5Yhivc).

If you just want a binary to flash on ESP, try  @srg74 repo for Wemos shield (https://github.com/srg74/WLED-wemos-shield/tree/master/resources/Firmware/WLED_wemos_shield).

tgrantjr, 2021-04-15:
ok i got the srg74 bin file and flashed it.  there is only sunrise.  Hopefuly i can figure out how to reverse it so i can do a sunset or even prolong it out so it stays on.

blazoncek, 2021-04-15:
As far as I know there are two triggers, one for sunrise and another for sunset.
[image]

If you trigger a playlist (a specially prepared preset) you can achieve your desired effect. One triggered at sunrise and another at sunset.

tgrantjr, 2021-04-16:
I got that checked and the time and everything set. How would I make it lights on in between the two when it will always be different times?  With those selected does it do that automatically?

blazoncek, 2021-04-16:
Please look at the playlist creation.
[https://github.com/Aircoookie/WLED/wiki/JSON-API#playlists](https://github.com/Aircoookie/WLED/wiki/JSON-API#playlists)

You trigger 1 playlist at sunrise and 1 at sunset.

Aircoookie, 2021-04-16:
Playlists not needed.
You can just make two presets, both with the "Sunrise" effect, but with different speed settings. Make sure your palette is set to default. Set the intensity to your liking (more = larger white part of the "sun")
- Speed 1-60 will make a sunrise over that amount of minutes and then stay on.
- Speed 60-120 will make a sunset over that amount minus 60 (so, for a 30 minute sunset, you would need a speed of 90), and then stay off.

Now you can set these 2 presets to either be triggered at a certain time or at the real sunrise/set :)

tgrantjr, 2021-04-18:
I made the presets but i can't figure out how to make it with the real time. I have wifi setup and the time automatically selected. I'm still confused sorry.

closed 2021-08-28 by stale[bot]