wled/WLED #1882: WS2812-E support & not remembering pin assignments

ThatBigPrint, 2021-04-11:
My bug is simply randomly losing some of the pin assignments and altering the led numbers per pin output, also glitching when driving more than one set of WS2812-E as I understand there is a reset timing issue similar to WS2812-C I need to fix this asap, for now, I suppose I will need to reinstall generic Chinese controller to get us by? I can recompile myself although I don know where to adjust the timing?

Please see below 

total LEDs 350

output 1 

Pin 32

Start 0 count 70

output 2 

Pin 33

Start 70 count 130

output 3

Pin 25

Start 130 count 175

output 4

Pin 26

Start 175 count 225

output 5 

Pin 27

Start 225 count 280

output 6 

Pin 14

Start 280 count 350

after the power cycle the numbers change but not all of the time I have tried 3 different boards now to no avail 
mainly the led numbers change for instance 

output 5 changes to start 225 and count 225

and 6 start 280 count 70 

and sometimes the pin numbers end up being null ie no contents 

and occasionally it will lose everything and return to defaults ie pin 2 etc 

i have checked flash and cleared it multiple times 

**Expected behavior**
To retain settings 

**WLED version**
 - Board: [ESP32 dev]
 - Version [latest precompiled binary for the esp32]
 - Format [Binary]

**Additional context**
Please help asap I have rather embarrassed myself making a boat sign for my friend 

Thank you for your help!

blazoncek, 2021-04-13:
Please post your cfg.json (available at http//wled-ip/edit) after a successful save.

ThatBigPrint, 2021-04-13:
Thanks for your prompt reply please find the text document attached (it wouldn't let me upload .json)

[cfg.txt](https://github.com/Aircoookie/WLED/files/6303204/cfg.txt)

blazoncek, 2021-04-13:
The important part:
```
  "hw":{
    "led":{
      "total":350,
      "maxpwr":850,
      "ledma":0,
      "rgbwm":3,
      "ins":[
        {
          "en":true,
          "start":0,
          "len":70,
          "pin":[
            32
          ],
          "order":0,
          "rev":false,
          "skip":0,
          "type":22
        },
        {
          "en":true,
          "start":70,
          "len":130,
          "pin":[
            33
          ],
          "order":0,
          "rev":false,
          "skip":0,
          "type":22
        },
        {
          "en":true,
          "start":130,
          "len":175,
          "pin":[
            14
          ],
          "order":0,
          "rev":false,
          "skip":0,
          "type":22
        },
        {
          "en":true,
          "start":175,
          "len":175,
          "pin":[
            13
          ],
          "order":0,
          "rev":false,
          "skip":0,
          "type":22
        },
        {
          "en":true,
          "start":225,
          "len":125,
          "pin":[
            25
          ],
          "order":0,
          "rev":false,
          "skip":0,
          "type":22
        },
        {
          "en":true,
          "start":280,
          "len":70,
          "pin":[
            26
          ],
          "order":0,
          "rev":false,
          "skip":0,
          "type":22
        }
      ]
    }
```
And I can see couple of errors:
- start of bus (strip) 2 is 70 and length is 130, that would give start 200 for bus 3
- start of bus 3 is 130 which is incorrect, should be 200
- and similarly all other buses
- the total count of leds does not corespond to actual sum of lengths

You are entering data in settings page incorrectly.

ThatBigPrint, 2021-04-13:
so output 2 should be 

output 2

Pin 33

Start 0 count 60 

?

Please note the orders and counts have been messed up by the bug they were all incremental as described at the beginning 
or is that caused by my wrong implementation?  

also, would this be causing the flickering erratic behaviour?   

and is my resistor that fixed the Chinese controllers issues causing problems here  ?

blazoncek, 2021-04-13:
Why would bus 2 start at 0? Aren't there any LED's on the bus 1? You continue with count from previous bus.

@Aircoookie you should seriously consider my implementation of LED settings.

ThatBigPrint, 2021-04-13:
so to clarify there are for instance 60 LEDs on the letter on bus 2 so, first letter on bus 1 would have 70 leds 

therefore below ?

output 1 (letter containing 70 LEDs)

Pin 32

Start 0 count 70

output 2 (letter containing 60 LEDs)

Pin 33

Start 70 count 60

output 3 (letter containing 45 LEDs)

Pin 25

Start 130 count 45

and the total of all LEDs driven by the esp combining all of the pinouts so in this example 175 would go into the total LEDs field? please confirm and apologies if I'm being slow I appreciate the assistance 

and just to confirm I can use any pins I want on the esp32? or is there any that I shouldn't?

blazoncek, 2021-04-13:
If you test-drive my experimental fork (look at my repo, dev branch) the values will be calculated automatically and you can get values from there.
Other than that, join the WLED Discord where you can get advice from other users.

ThatBigPrint, 2021-04-13:
was I on the correct track? I'm having trouble compiling your version

blazoncek, 2021-04-14:
@srg74 has a precompiled version in his Wemos shield repo. You can try that.

ThatBigPrint, 2021-04-14:
Good news, sorta, your version is far more intuitive to configure multiple outputs! and settings are being retained 

and the lights are working but there is a slight issue still the last letter on pin 23 is randomly flickering, I tried adding a 310r resistor and that made it worse please see the clip below 

this is without a resistor 
[https://youtu.be/s98xt_VVI6Q](url)

this is with one 310r
[https://youtu.be/1dydJhj4FFU](url)

please note the cable length is only about 15 inches I'm not sure if it's signal reflections again or slightly off timing for the ws2813e LEDs I'm using

blazoncek, 2021-04-15:
It may be a grounding issue or a bad pixel. Also check if you need additional power injection at the end.
Other than that, why did you choose multipin for that? You could go with a single pin and multi segment.

ThatBigPrint, 2021-04-15:
power injection I think I have pretty much ruled out tbh thick copper bus bar between letters and the PCBs have 2 full layer power planes... I chose multi-pin because at the time I didn't know much about wled so I designed it for that Chinese controller. and the issue with that is it's machined and installed now I will be doing things differently next time. 

the issue look to flicker when it's being adjusted? ie when it's on a solid colour say red and you change it to say white it changes over but flickers slightly when doing so? should I put a lower or higher value resistor?

there is a capacitor array inside each letter to help stabilize power it's pretty much what the Chinese controller did but this time the resistor didn't work? 

am I totally mistaken regarding the reset timing on my led? and the others are just on the edge of working?

blazoncek, 2021-04-15:
Observe closely, if the whole strip flickers then try adjusting resistor and/or add level shifter (if not already in place), if only part of the strip flickers then ...
I'd say a faulty pixel then. Had those screw up my installs.

ThatBigPrint, 2021-04-15:
What resistor values do you have success with 310r range?

blazoncek, 2021-04-15:
I never used any but I've seen boards with 56Ohm, 62Ohm and up to 320Ohm resistors.

ThatBigPrint, 2021-04-15:
Just as a form of closure, all worked out in the end, 76r resistor sorted the flicker out by the looks of things; I can't thank you guys enough for your support. 

Keep up the brilliant work!

closed 2021-04-15 by ThatBigPrint