wled/WLED #1884: Brightness when controlling via DDP

mase76, 2021-04-11:
Hi!
When I control the device via DDP with xLights, the brightness-slider in WLED should not have any effect. The general brightness control should only be taken by the controlling protocol, and WLED should be at full brightness. Only the brightness limiter should be active.
The current behaviour: I select a solid light with the lowest brightness as a power-on-indicator. When xLights takes control, the brightness settings of WLED and xLights are somehow merged together. This results in dark or even off leds.

Aircoookie, 2021-04-11:
Hi!
This is intended behavior.
If you would like to control the brightness exclusively from xLights, you can enable `Force max brightness` in the Realtime section of Sync Settings.

mase76, 2021-04-11:
I see. Is the limiter still active then, to prevent from too high current?

Aircoookie, 2021-04-11:
Yes! The target brightness is set to maximum, but actual output still limited if required.

mase76, 2021-04-11:
Great! Thx!

closed 2021-04-11 by mase76