wled/WLED #1885: Cycling through presets using HTTP API

gorstj, 2021-04-11:
**Describe the bug**
Using HTTP API to cycle through presets using &PL=~ stops at last preset

**To Reproduce**
use local version of command http://IP-Address/win&PL=~
Keep activating until you get to last preset
In web interface you will see an error 'Error 12. The selected preset does not exist'

**Expected behavior**
Expect it to go around to Preset 1 rather than get stuck on last preset and give error message

The wiki page suggests that this is the expected behaviour:

https://github.com/Aircoookie/WLED/wiki/HTTP-request-API#in-decrementing-values
For example, use PL=~ to go to the next preset. Using just ~ without a number will increase the value by 1, ~- will decrease it by 1. The value will then wrap around, so using A=~- when A is 0 will set A to 255.

**WLED version**
 - Board: Wemos D1 Mini ESp 8266
 - Version 0.012.0-b4
 - Format Binary

Aircoookie, 2021-08-19:
Hey!
You can set the range of presets with the `P1` and `P2` commands, e.g. to cycle from preset 5 all the way through 12 you can use `win&P1=5&P2=12&PL=~`. All presets should exist to avoid the error you mentioned.

I will reformat this issue as an enhancement request to skip non-existant presets within the cycle range.

blazoncek, 2022-10-10:
Also possible via JSON API (although not yet documented) as of 0.13

closed 2022-10-10 by blazoncek