wled/WLED #1888: Enigmalight - UDP

DavidAntonin, 2021-04-12:
Hello, I am using WLED together with my Enigma2 box with enigmlight plugin. It is possible to set up WLED to automatic turn ON/OFF leds with/without signal from enigmalight? It is using UDP/NodeMCU.
Thanks

pierrecinema, 2021-04-12:
Hello . I'm using the same config. Without trim switch the wled over Google home/openhab ore over Philipps hue

schwatter, 2021-04-13:
Hello,

i have built that feature in. Take a look here.

https://www.boxpirates.to/index.php?thread/206406-plugin-enigmalight-ambilight-f%C3%BCr-enigma2-receiver-oe1-6-oe2-0-oe2-2-oe2-5-dreamb/&pageNo=20

Newest version is in the thread. EnigmaLightWledTest24.rar

Cheers schwatter

DavidAntonin, 2021-04-13:
> Hello,
> 
> i have built that feature in. Take a look here.
> 
> https://www.boxpirates.to/index.php?thread/206406-plugin-enigmalight-ambilight-f%C3%BCr-enigma2-receiver-oe1-6-oe2-0-oe2-2-oe2-5-dreamb/&pageNo=20
> 
> Newest version is in the thread. EnigmaLightWledTest24.rar
> 
> Cheers schwatter

Can you please explain it more detailed? I cannot find EnigmaLightWledTest24.rar

pierrecinema, 2021-04-14:
?

https://www.boxpirates.to/index.php?thread/206406-plugin-enigmalight-ambilight-f%C3%BCr-enigma2-receiver-oe1-6-oe2-0-oe2-2-oe2-5-dreamb/&pageNo=20

schwatter, 2021-04-14:
1. Save your EnigmaLight\profiles folder
2. https://www.boxpirates.to/index.php?attachment/74002-enigmalightwledtest24-rar
3. Just copy and paste Enigmalight from rar to folder /usr/lib/enigma2/python/Plugins/Extensions/EnigmaLight
4. Copy your EnigmaLight\profiles folder back
5. Restart
6. When you use WLED with usb, go in Enigmalight to settings and edit the ip to yours
7. If you use WLED over IP only with the script from StefanRu1, Enigmalight now extract all IP's from enigmalight.conf automatic
8. Mixed setup is also ok (IP and USB)
8. In settings you can also change the state for WLED at start/stop/sleep

cheers schwatter

DavidAntonin, 2021-04-17:
schwatter: Thank you, I found StefanRu1 scripts: https://github.com/StefanRu1/WLED-Enigmalight/tree/main/Enigmalight%20Plugin. Colors from WLED android app are acurate but from enigmalight not

closed 2021-08-28 by stale[bot]