wled/WLED #1892: How to use WS2815

LukasVFL99, 2021-04-14:
Hey how do I need to Setup the LED Preferences if I use a WS2815 I cant get the WW working.
Its a RGBWW Led Stripe. Any help would be appreciate.
Thanks. :)

blazoncek, 2021-04-14:
AFAIK WS2815 is a RGB (GRB order) strip and not RGBW(W).
SK6812 is RGBW.

LukasVFL99, 2021-04-15:
@blazoncek I ordered it from AliExpress and there they said it's rgbww

blazoncek, 2021-04-15:
So, if I say "throw it away", will you? I guess not. ;) Don't belive everything that is said.

If it is WS2815 it will be 12V and will have DI/BI and DO/BO signal lines.
[image]

If it is RGBW it will have a separate LED for white.
[image]

LukasVFL99, 2021-04-15:
Ah damn it. Yeah I guess I will deal with it but thanks :)

closed 2021-08-19 by Aircoookie