wled/WLED #1895: 5050 led strip support

ericksvasc, 2021-04-14:
sorry for my English, but how do I use the Wled with 5050 led strip

ericksvasc, 2021-04-15:
> sorry for my English, but how do I use the Wled with RGB 5050 led strip

sansillusion, 2021-04-15:
In settings/leds you can choose the led type and pins(s).
If your question is about how to connect your leds to your board I recommend doing a search with your board type/model + 5050 in your language. That should give you good results that you can apply and then configure wled to use the pins chosen.

srg74, 2021-04-17:
Answered in #1897

ericksvasc, 2021-04-19:
> In settings/leds you can choose the led type and pin(s).
> If your question is about how to connect your leds to your board I recommend doing a search with your board type/model + 5050 in your language. That should give you good results that you can apply and then configure wled to use the pins chosen.

already researched in my language and did not find it. do you know of any tutorials in English?

srg74, 2021-04-19:
https://www.instructables.com/Driving-RGB-LED-strips-off-an-Arduino/

closed 2021-08-28 by stale[bot]