wled/WLED #1896: WS2811 Flickering around 70% and above brightness.

Doyle4, 2021-04-14:
I have built a giant infinity mirror for a bar, in total it has 1410 LEDs (470 inputted to wled config due to X3 leds as one).
The power inline has been jumped from the start of LED strip to the end as the end meets to the start and a 12V 100amp PSU is powering the strips. 
X2 5meter with 15 leds removed from each strip, one strip at the front and another strip at the back running in opposite direction.
Using a Node MCU flashed with downloadable 0.11 WLED.

Issue it is having is, some effects look glitchy, some effects will crash the Node and a reboot is needed.
Id say 90% of the effects work great, even at 100% brightness, but a couple of motion effects and Solid all glitch out, or major flickering.

How can I resolve this please?

Thanks.

huggy-d1, 2021-04-14:
Are you using a level shifter to boost the NodeMCU 3.3V digital output signal to 5V the LED strip expects?  If you are using a level shifter, that's great and ignore my question.

Now on to the real likely problem.

Anytime I hear about flickering at higher brightness levels, I immediately jump to suggesting adding more power injection points, increase wire size from the supply, or increase the number of wires from the supply to the strip.

For example, for 3x LED strips, you would have something like this:

```
P/S -----o---------[STRIP1 START]------[2m]-----[2m]------[STRIP1 END]
              o---------------------------------^            ^                   ^
              o-------------------------------------------+                   +
              o----------------------------------------------------------+
              o---------[STRIP2 START]------[2m]-----[2m]------[STRIP2 END]
              o---------------------------------^            ^                   ^
              o-------------------------------------------+                   +
              o----------------------------------------------------------+
              o---------[STRIP3 START]------[2m]-----[2m]------[STRIP3 END]
              o---------------------------------^            ^                   ^
              o-------------------------------------------+                   +
              o----------------------------------------------------------+
```
That's 12 terminals off the power supply using a distribution block of some sort.  Or 3 distribution blocks from the supply to each strip with 4 terminals out.
Try it with just one strip and see how it goes.  You can measure the voltage along the strip, mid-way along each strip at a +V solder pad.  Compare it to the start strip voltage.  If it is drooping, then you need more power injection.

Doyle4, 2021-04-15:
@huggy-d1 
Thank you very much for all the information!

I missed the level shifter, I have ordered some and hopefully be here soon and fingers crossed this resolves it.
Im going to check for voltage drop today and finish the wiring whilst I wait for the LS.

Thanks again!

Doyle4, 2021-04-16:
I used the single pixel trick to test and flickering stopped :)
Level shifters are on the way too.

Thanks again.

closed 2021-04-16 by Doyle4