wled/WLED #1898: simple light effect

DanteHA, 2021-04-15:
Hi Aircoookie, 

The WLED has the light effect that when this effect is selected, the tape starts all off and the light begins to travel (keeping the LEDs on where it has already passed) and when it reaches the end of the tape it remains all lit (as if it were the effect solid).

If so, would you be able to tell me which one?
if not, would it be too much to ask to have this effect added?

(I have ws2812b on every edge of my ceiling and I would like to use this as an ambient light, of course, keeping the color selection options brightness intensity)

I love WLED, I already ordered some more nodemcu to change my old led controllers to WLED!

Thanks from Brazil

Aircoookie, 2021-04-16:
Hi,

a very similar effect to what you would like is `Wipe`. It turns on the LEDs one after the next, but when it is done, it will turn off the LEDs again in the same order. There is no effect supported that stays on the lit position afterwards, but there is a workaround using the new playlist feature to get your desired result, however it is still quite complicated to use.

You could make a preset with the Wipe effect set up with the color/speed you like, then refresh the page. Now you can open the preset and send me the contents of the "API command" text box, and I will get back to you with some steps to make it stay on after one wipe :)

I'm very glad that you like WLED so much!

baerni, 2021-04-16:
I am also interested in this effect. I would like the light to turn on gradually from one side to the other (like the Wipe effect), but when all the LEDs are on, it should be on all the time. 
On the other hand, if I would like to turn off the light, it would also turn off gradually from one side to the other.
I would like to have an effect like [here](https://www.youtube.com/watch?v=G72JtVlssW4) or [here](https://www.youtube.com/watch?v=NHkju5ncC4A). Is it possible with WLED and macro or API etc?

ddijanic, 2021-04-20:
> I am also interested in this effect. I would like the light to turn on gradually from one side to the other (like the Wipe effect), but when all the LEDs are on, it should be on all the time.
> On the [...]

look at this for the stairs
https://github.com/Aircoookie/WLED/tree/master/usermods/Animated_Staircase

sssushi, 2021-04-21:
Hello,

i am trying to use the dissolve effect followed by solid for my mirror light. Is there a way to trigger the dissolve from the beginning so the light will „appear“ ?

utech-git, 2021-08-30:
> Hi,
>
> a very similar effect to what you would like is `Wipe`. It turns on the LEDs one after the next, but when it is done, it will turn off the LEDs again in the same order. There is no effect [...]

Hello, Aircookie.

I want the exact same effect for my Wled. I have a LED strip that climbs up my wall then runs into the ceiling, and I want this effect so when I turn it on, it "wipes on" bottom to ceiling like a "lightsaber", than stays on 

I tried the "Animated Staircase" usermod but didn't work very well. 

So I ran into this thread. I did what you asked and here is the API command:

``
{"on":true,"bri":128,"transition":7,"mainseg":0,"seg":[{"id":0,"start":0,"stop":60,"grp":1,"spc":0,"of":0,"on":true,"bri":255,"col":[[255,160,0],[0,0,0],[0,0,0]],"fx":3,"sx":233,"ix":128,"pal":0,"sel":true,"rev":false,"mi":false},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0}]}
``

I tried changing these "stop" parameters at the end but didn't work. 

Thanks!

blazoncek, 2021-08-30:
Have you tried Tetrix?

Messkid21, 2022-02-14:
Hi ! i'm a huge fan of WLED and i'm trying to include this kind of effect on a project : I have 6 segments and I want to trigger a Wipe on a segment every X second (each segment will have the effect but not on the same time).

It's  really complicated from the UI, I tried with the playlist but if the solid effect is not exactly a multiple of the cycle time it's all messed up... any news from a way to achieve this ?

Thanks a lot !

radu022003, 2023-02-03:
I would like to have exact the same behavior for my lights.
I am using the wled with HA, but how can we turn the lights on directly with the playlist?

thank you and best regards

1liminal1, 2023-04-16:
What did you get working? And how?

blazoncek, 2023-04-16:
Please use WLED [forum](https://wled.discourse.group) or Discord for help and support questions.
Closing as it is possible to achieve OP request using existing effects in combination with playlist.

closed 2023-04-16 by blazoncek