wled/WLED #1899: Add support for toggle wall switches

forerunnert, 2021-04-15:
I have retrofitted a ceiling light with LED strips and connected the existing wall switch through an opto coupler.
My problem is that there is WLED support for buttons (on/off - on/off - on/off cycle) and my wall switch is of the toggle kind (on  - off - on cycle).

I'd be great if there would be an option that maps a toggle switch to a button switch, so:

button state change | toggle switch state change | event
------------ | ------------- | -------------
on/off | off->on or on->off | short press
on/off - on/off | off->on->off or on->off->on | double press
on - off | off->on------>off or on->off------>on | long press

I'm currently using ESPHome but I've found WLED to be more fantastic for LED applications.
Alternatively exposing the button events in WLED to Home Assistant could let me control the same through HA, but I'd rather have the functionality run on the ESP independently.

I imagine retrofitting lights and wall switches to be a common user scenario.

Aircoookie, 2021-04-16:
Hi! I've recently added Toggle Switch support, it is not yet part of v0.12.0, but available in latest master and will be part of v0.12.1!
It executes the short press macro on off->on and the long press macro on on->off, so it doesn't yet have the full feature set you are proposing, but it's a start :)

forerunnert, 2021-04-16:
Great! I'll give it a try

closed 2021-04-16 by forerunnert

utech-git, 2021-06-01:
Hello, there.

Did you manage to get the "switch" mode working? I'm having a hard time getting it correct on [version 12.1-b1](https://github.com/Aircoookie/WLED/releases/tag/v0.12.1-b1)

Right [here](https://github.com/Aircoookie/WLED/blob/master/CHANGELOG.md#build-2104120) it states what @Aircoookie explained, but I couldn't get it to work.

I have a Sonoff Basic flashed with WLED. I am using GPIO3 (RX pin) as the button/switch pin (the same pin I use for momentary switch when using Sonoff Basic with Tasmota). 

First I created a Preset named Toggle with API T=2. Then I set both Short Press Macro and Long Press Macro to execute this preset (Preset 1)
But with this setup, when shorting GPIO3 to GND it behaves as a button.

Am I doing anything wrong?

The expected behaviour I wanted is: When I short GPIO3 go GND it toggles the state. When I separate GPIO3 and GND, it toggles the state again.

The behaviour I am getting is: When I short GPIO3 to GND it toggles the state. When I separate GPIO3 and GND nothing happens.

(I dont think the problem is with the pin I am using because I get the same behaviour using GPIO0 - the pushbutton present on Sonoff Basic)

Thanks