wled/WLED #1900: WLED vers DMX ?

arnaud-aub, 2021-04-15:
Bonjour j'ai un sonoff avec wled dessus et j'aimerais savoir s'il est possible de contrôler du DMX avec ? J'ai une puse MAX-485 pour potentiellement convertir en DMX, si je branche la pin de sortie sur la donnée de celle-ci, cela va fonctionner ? Merci

blazoncek, 2021-04-16:
English? Please.

arnaud-aub, 2021-04-16:
Hello I have a sonoff with wled on it and I would like to know if it is possible to control DMX with it? I have a MAX-485 puse to potentially convert to DMX, if I plug the output pin into the data from it, will that work? Thank you

Aircoookie, 2021-04-16:
Yes, this is possible if you do a custom compile with the `-D WLED_ENABLE_DMX` build flag enabled :) https://github.com/Aircoookie/WLED/wiki/Compiling-WLED#installation-guide-platformio-recommended

arnaud-aub, 2021-04-16:
hi i cant seem to do what is begged i dont understand, can someone help me please. I have a sonoff with the only pin 14 available and I want to do DMX, what should I do?

Aircoookie, 2021-04-16:
Hi, please try this binary, it has DMX compiled in on pin 14 and it should be possible to attach MAX-485 to it.
https://drive.google.com/file/d/1bzT2ww7PeslKSB39nV-D9TbSMOTKyK8m/view?usp=sharing

arnaud-aub, 2021-04-16:
hi i put the binary on the sonoff, enabled the DMX on the WLED interface plugged into the MAX-485 but it doesn't work, can I have some help please.

closed 2021-08-28 by stale[bot]