wled/WLED #1901: Color correction for UDP signals

ahmed-shehata, 2021-04-15:
**Is your feature request related to a problem? Please describe.**
Hyperion sends color data to WLED instance over UDP. LED strips are not all created alike, they differ in quality and color calibration. It would be very useful if there was a way to calibrate and configure Saturation/Brightness of the incoming colors.

**Describe the solution you'd like**
The proposed solution would be a modifier option in settings to modify the factor of saturation/brightness to increase/decrease (1x,1.5x, etc..) 

**Describe alternatives you've considered**
I have built a solution as a proxy UDP server (Hyperion -> UDP Proxy server (modifies color) -> WLED) and it works fine, I would like to implement it in WLED for better latency and easier configuration.

**Additional context**
[image]

I am happy to implement this myself, I would love some code pointers on how to:
- Add and save a new setting option (Sync interfaces settings)
- Load the saved setting option (saturation/brightness factors) and apply them in [udp.cpp](https://github.com/Aircoookie/WLED/blob/master/wled00/udp.cpp#L380-L392)

ahmed-shehata, 2021-04-16:
Pull request here: https://github.com/Aircoookie/WLED/pull/1902

Happy to hear your thoughts, @Aircoookie.

moustafa-nagaty, 2021-04-24:
Thats a great idea, will definitely be useful!

ReviloEgros, 2021-08-09:
Would love to see this implemented!

PR #1902 (2021-08-19): UDP Signal color correction
Closes #1901

Before: (very dim)
[image]

After: (vivid)
[image]
closed 2021-08-19 by Aircoookie