wled/WLED #1903: WiFi not working on ESP32?

ThiRok, 2021-04-16:
**Describe the bug**
A clear and concise description of what the bug is. Please quickly search existing issues first!

**To Reproduce**
Steps to reproduce the behavior, if consistently possible

**Expected behavior**
A clear and concise description of what you expected to happen.

**WLED version**
 - Board: [e.g. Wemos D1, ESP32 dev]
 - Version [e.g. 0.10.0, dev200603]
 - Format [e.g. Binary, self-compiled]

**Additional context**
Anything else you'd like to say about the problem?

Thank you for your help!

mikey0000, 2021-04-19:
It's crashing for me on nodemcu esp32.

```
 Guru Meditation Error: Core  panic'ed (Interrupt wdt timeout on CPU0)
 Core 0 register dump:
 PC       0x40087806  PS       0x00060b4  A0       0x8008ba  A1       0x3fab0  
 A2       0x3fb0b  A3       0x3fb608  A4       0x00000000  A5       0x0000001a 
 A6       0x400844c  A7       0x3fc2b  A8       0x3f24cc A9       0x3fab0  
 A10      0x00000010  A11      0x00000000  A12      0x3fb344  A13      0x3f73c4  
 A14      0x3f74270  A15      0x3fbfc SAR      0x0000001c EXCCAUSE 0x00000005  
 EXCVADDR 0x00000000  LBEG     0x4000ce  LEND     0x4000cf  LCOUNT   0xffff  
 
 ELF file SHA256: 000000000000000
 
 Backtrace:0x40087806:0x3fab00x4008ba:0x3fab0
 
 ore 1 register dump:
 PC       0x4000c71  PS       0x00060034  A0       0x8008ba  A1       0x3fbe0  
 A2       0x3fb058  A3       0x3f31c8  A4       0x00000014  A5       0x3fc250  
 A6       0x3fc298  A7       0x00000001  A8       0x00000001  A9       0x3f31c9  
 A10      0x000000a  A11      0x3fc718  A12      0x8008c63  A13      0x3fc220  
 A14      0x00000008  A15      0x00000001  SAR      0x00000016  EXCCAUSE 0x00000005  
 EXCVADDR 0x00000000  LBEG     0x400f0d  LEND     0x400f0f  LCOUNT   0x0000000b 
 
 ELF file SHA256: 000000000000000
 
 Backtrace:0x4000c71:0x3fbe00x4008ba:0x3fbe00x4008d54:0x3fbe00x4008d0a0x3fbfc
 
 Rebooting...
```

raddi, 2021-04-19:
Same here, AP is not showing up, ESP32 (also ESP32-WROOM) crashes.

mikey0000, 2021-04-19:
I built master targeting nodemcu32s and its stable and running fine, so might have to add another platform @Aircoookie

mikey0000, 2021-04-20:
Have a new/same backtrace on the master branch compiled firmware.
```

 Guru Meditation Error: Core  panic'ed (Interrupt wdt timeout on CPU0)
 Guru Meditation Error: Core  panic'ed (Unhandled debug exception)
 ebug exception reason: tack canary watchpoint triggered (DLE0 
 Core 0 register dump:
 PC       0x4000ce  PS       0x00060036  A0       0x80089364  A1       0x3fb040  
 A2       0x3fb058  A3       0x3fce0  A4       0x00000060  A5       0x3fb058  
 A6       0x3fc85c A7       0x3fc868  A8       0x00000000  A9       0x3fb380  
 A10      0x00000000  A11      0x00000000  A12      0x00060023  A13      0x3fb9c  
 A14      0x00000002  A15      0x3fb360  SAR      0x0000001c EXCCAUSE 0x00000001  
 EXCVADDR 0x00000000  LBEG     0x4000ce  LEND     0x4000cf  LCOUNT   0x00000005  
 
 ELF file SHA256: 000000000000000
 
 Backtrace:0x4000ce:0x3fb0400x40089361:0x3fb050 0x400894e:0x3fb120 0x40089740:0x3fb1400x4008592e0x3fb180
 
 ore 1 register dump:
 PC       0x4000c71  PS       0x00060034  A0       0x8008ba  A1       0x3fbe0  
 A2       0x3fb058  A3       0x3f32170  A4       0x00000014  A5       0x3fc240  
 A6       0x3fc288  A7       0x00000001  A8       0x00000001  A9       0x3f32171  
 A10      0x000000a  A11      0x00060a3  A12      0x8008c63  A13      0x3fc210  
 A14      0x00000008  A15      0x00000001  SAR      0x00000016  EXCCAUSE 0x00000005  
 EXCVADDR 0x00000000  LBEG     0x4000ce  LEND     0x4000cf  LCOUNT   0xffff  
 
 ELF file SHA256: 000000000000000
 
 Backtrace:0x4000c71:0x3fbe00x4008ba:0x3fbe00x4008d54:0x3fbe00x4008d0a0x3fbfc
 
 Rebooting...
```

mikey0000, 2021-04-20:
Odd, reflashed and its stable, I've changed the default pin to something other than 2

mikey0000, 2021-04-20:
Left it on and it died again this time here(master branch) 

  ```
Decoding stack results
 0x40086487: wifi_bt_common_module_disable at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/driver/periph_ctrl.c line 275
```

blazoncek, 2021-04-21:
wdt is a watch-dog-timer
your esp is doing something and not yield()-ing.
check your overall network traffic and try to disable all nonessential IO (MQTT, Alexa, ...) if any of them are enabled. also try to use different GPIO

sblantipodi, 2021-04-22:
Esp32 is a broken platform. Powerful but broken, just use esp8266

mikey0000, 2021-04-22:
> Esp32 is a broken platform. Powerful but broken, just use esp8266

Mind backing that up with some examples please. What's broken?

Esp32 should be supported by the company who made it just like 8266 surely its in their interest to fix the issues?

Is it the hardware itself that's broken or just the core software?

blazoncek, 2021-04-22:
> Is it the hardware itself that's broken or just the core software?

It is the core libraries that are subpar.

mikey0000, 2021-04-23:
Took your advice and used an esp8266 I fortunately had lying around. Stable as now. Shame about the esp32 being so buggy. I tried disabling everything (alexa etc) made no difference also seemed to be if I drew too much power and had a brown out it soft bricked the device till next flash.

ThiRok, 2021-04-23:
V0.11 is working like a charm on an ESP32 here. It is just V0.12 that give the problems here. Even an OTA update from a working ESP32 from 0.11 to 0.12 breaks the software. 

[image]

mikey0000, 2021-04-23:
I suspect its the updated core library for the esp32 that's caused the issue. Will downgrade it and see.

blazoncek, 2021-04-23:
esp32 libraries were updated to v3 in 0.12.

mikey0000, 2021-04-23:
Yeah I looked through the commit history.

kopierschnitte, 2021-04-26:
Strange. I had problems with a NodeMCU 8266 using 0.12.0 (device dropped Wifi after a few minutes/hours). Now using an ESP32 (M5Stack Atom) and it works great...

mikey0000, 2021-04-26:
@kopierschnitte what firmware/bootloader did you put on the esp32?

kopierschnitte, 2021-04-27:
@mikey0000 Just stock binaries from this repository. I used the Espressif flash to burn them.

mikey0000, 2021-04-27:
I've narrowed down the issue specifically to the WiFi AP in initAP. I don't know enough about the arduino WiFi libs/platformio to move much further other than it seems to be the same issue to do with marking a function with IRAM_ATTR. I'll try and see if I can work it out when I'm not so tired.

sblantipodi, 2021-04-27:
esp32 is simply not stable enough, its TCP stack never grew up compared to ESP8266.
simply avoid that chip for WiFi things...

closed 2021-09-07 by stale[bot]