wled/WLED #1904: Flashing New Release

crosby000, 2021-04-17:
I have made several attempt to flashing wled 0.12 firmware onto esp32 and esp32wroom
to no success for some reason. and have bricked some of them and not able to reflash those boards
is there something i am missing when attempt to flash wled to the esp. that once it has successfully says flashing was completed that it should not be accessed only one time i was able to connect to wled-ap but never saw the main page. 
need your help 
thank you

**Take a look at the wiki and FAQ, perhaps your question is already answered!**
[FAQ](https://github.com/Aircoookie/WLED/wiki/FAQ)

**Please consider asking your question on the WLED forum or Discord**
[Forum](https://wled.discourse.group/)
[Discord](https://discord.gg/KuqP7NE)
[What to post where?](https://github.com/Aircoookie/WLED/issues/658)

**If you do not like to use these platforms, delete this template and ask away!**
Please keep in mind though that the issue section is generally not the preferred place for general questions.

srg74, 2021-04-17:
Please visit WLED discord for quick answers.

huggy-d1, 2021-04-22:
Did you find answers either in the WLED discord or wiki where it mentions how to flash ESP32 devices?

closed 2021-08-22 by Aircoookie