wled/WLED #1909: Razer Chroma Support

AlexCruz02, 2021-04-21:
I know this will sound complicated but could someone add Razer Chroma support because it is easy to sync with all your razer products or recommend a repo for that.

TheFreakPlays, 2021-04-22:
That would be nice

AriaFantom, 2024-11-16:
Yes i hope it gets added

netmindz, 2024-12-05:
Are you wanting to add WLED to your Razor Chroma setup or drive your Razor Chroma hardware from WLED?

AriaFantom, 2024-12-05:
> Are you wanting to add WLED to your Razor Chroma setup or drive your Razor Chroma hardware from WLED?

We want razer chroma to control wled, you can see razer chroma app have option to control other brand leds like that we want

netmindz, 2024-12-05:
Then you need to contact them. We can't add features to their app. However if that are making a large markup on their hardware it's not really in their interest to let you use your own cheap hardware.

Alternatively you would need to reverse engineer their protocol so you could emulate their hardware. Once that is done then reopen the issue

AriaFantom, 2024-12-07:
> Then you need to contact them. We can't add features to their app. However if that are making a large markup on their hardware it's not really in their interest to let you use your own cheap [...]

What if we want reverse

netmindz, 2024-12-08:
The same is true, you would need to reverse engineer their protocol

AriaFantom, 2024-12-08:
If we could why would ask here?

netmindz, 2024-12-08:
The core WLED development team are unlikely to invest their own personal time into such a project.

You may find other users who also own this hardware and therefore might be motivated. Such discussion is better on discord