wled/WLED #1910: Calling a preset from HTTP API seem not to apply Segment immediately

swampdogmash, 2021-04-21:
**Describe the bug**
I have a Wemos D1 Mini board running WLED 0.12.0. I call the board using curl from within .cmd files. When switching from a preset with one segment of solid colour covering all LEDs to a preset 2 separate solid colours with 2 segments defined, all LEDs light up the colour of segment one of the second preset on the first API call and change to the correct preset after a second identical call. 181 ws2812s are connected. 

**To Reproduce**
2 presets at position 1 and 12

position 1
`{"on":true,"bri":255,"transition":7,"mainseg":0,"seg":[{"id":0,"start":0,"stop":103,"grp":1,"spc":0,"on":true,"bri":255,"col":[[8,255,0],[190,115,255],[0,0,0]],"fx":0,"sx":3,"ix":255,"pal":0,"sel":false,"rev":false,"mi":false},{"id":1,"start":103,"stop":181,"grp":1,"spc":0,"on":true,"bri":255,"col":[[0,0,255],[0,0,0],[0,0,0]],"fx":0,"sx":128,"ix":128,"pal":0,"sel":true,"rev":true,"mi":false},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0}]}`

position 12
`{"on":true,"bri":255,"transition":7,"mainseg":0,"seg":[{"id":0,"start":0,"stop":181,"grp":1,"spc":0,"on":true,"bri":255,"col":[[255,255,255],[190,115,255],[0,0,0]],"fx":0,"sx":3,"ix":255,"pal":0,"sel":true,"rev":false,"mi":false},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0}]}`

call position 12 with a script:

start /b cmd /C curl "http://10.128.0.11/win&PL=12" > nul
all LEDs go white and one segment is seen in the config after refreshing the web interface

then call position 1 with a call:
start /b cmd /C curl "http://10.128.0.11/win&PL=1" > nul
All leds go green - refreshing the interface shows that 2 segments are now defined but not the different colours
 saving the config at this point to a preset shows that the colour seems not to have updated:
`{"on":true,"bri":255,"transition":7,"mainseg":0,"seg":[{"id":0,"start":0,"stop":103,"grp":1,"spc":0,"on":true,"bri":255,"col":[[8,255,0],[190,115,255],[0,0,0]],"fx":0,"sx":3,"ix":255,"pal":0,"sel":false,"rev":false,"mi":false},{"id":1,"start":103,"stop":181,"grp":1,"spc":0,"on":true,"bri":255,"col":[_**[8,255,0**_],[0,0,0],[0,0,0]],"fx":0,"sx":128,"ix":128,"pal":0,"sel":true,"rev":true,"mi":false},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0}]}`

call position 1 again
start /b cmd /C curl "http://10.128.0.11/win&PL=1" > nul
segments 1 goes to green, segment 2 goes to blue

**Expected behavior**
calling position 12 should light up all LEDs white. Calling position 1 should change the second section of leds to blue and the first section to green.

**WLED version**
 - Wemos D1 mini
 - Version 0.12.0
 - Format binary from release build full 8266 version

Thank you for your help!

huggy-d1, 2021-04-22:
When you say position 1 and 12, do you mean segment 1 and 12, or do you mean preset 1 and preset 12?
I hope you mean preset 1 and preset 12.
If you select preset 1 from WLED web or smartphone app, do you get the expected results?  If you select segment 1, do you see _anything_ change?

swampdogmash, 2021-04-23:
I did indeed mean preset 1 and 12. I do get the expected result from pressing the preset button in the web gui.

swampdogmash, 2021-09-05:
This still seems to be an issue, is anyone able to replicate it at all?

blazoncek, 2021-09-05:
Did you try the latest beta?

swampdogmash, 2021-09-06:
Just installed 13b2 and it seems to have fixed the issue :)  thank you :)

blazoncek, 2021-09-06:
> Just installed 13b2 and it seems to have fixed the issue :) thank you :)

If you get random LED flickering it is a known issue and will be resolved in next release.

closed 2022-04-16 by Aircoookie