wled/WLED #1913: Hardware profiles for pin assignment

tbnobody, 2021-04-23:
Currently every user has to adjust each pin assignment by their own and for their specific hardware. If WLED is running on commercial or self designed PCB's it would be be nice to have profiles (e.g. in json format) to specify all the different hardware pins (IR/Ethernet/Data/Clk etc.) and just choose the specific profile within the webinterface to assign all the correct pins.

blazoncek, 2021-04-29:
This can be achieved by HW vendor by pre-compiling with certain defaults set and providing a binary for users.
All supported HW devices (button, IR, relay, LED pins) are available to be configured in this way.

CWempe, 2021-05-04:
I think this is a good idea.

For example when you use the [WLED-wemos-shield](https://github.com/srg74/WLED-wemos-shield).
Or any other hardware that has a wither "market share".

Tasmota solved this with [templates](https://tasmota.github.io/docs/Templates/).
You can either select a template from a long list of a drop-down menu in the web interface.
Or you can import/export a custom template (JSON format)and share this with others or reuse this for your similar projects.

blazoncek, 2021-05-04:
It is possible to copy cfg.json from one device to the other which would give you a similar experience.
Just reboot after uploading (and possibly change device name if you need).

henrygab, 2021-07-05:
This is a **_GREAT_** idea.  

<details><summary>Different problem than PinManager class solves...</summary><P/>

The [`PinManager`](https://github.com/Aircoookie/WLED/blob/8c44147a4596112a7f5d5720ae809949399e9d44/wled00/pin_manager.cpp#L13) class is intended to prevent use one pin for multiple purposes (or both input and output, or an input-only pin for output, ...).  Two downsides:

1. Every PR that uses pins needs to know about and properly use this class.  This fails over time, causing [really hard-to-diagnose problems](https://github.com/Aircoookie/WLED/issues/1027#issuecomment-723481641).
2. `PinManager` doesn't provide a method to define board / product specific availability of pins
3. `PinManager` doesn't provide an **_extensible_** method to determine (in code) how many of a given functional thing exists

Here's a small list of potential pin uses ...

| Max Pins | UseType | Use `PinManager` |
|------|------|------|
| 1 | IR | [Yes](https://github.com/Aircoookie/WLED/blob/8c44147a4596112a7f5d5720ae809949399e9d44/wled00/cfg.cpp#L170-L174)
| 1 | Relay | [Yes](https://github.com/Aircoookie/WLED/blob/8c44147a4596112a7f5d5720ae809949399e9d44/wled00/cfg.cpp#L181-L186) |
| 4 | Button | [Yes](https://github.com/Aircoookie/WLED/blob/8c44147a4596112a7f5d5720ae809949399e9d44/wled00/cfg.cpp#L134-L139) |
| 4 | Ethernet | **_No(!)_**, see [1](https://github.com/Aircoookie/WLED/blob/8c44147a4596112a7f5d5720ae809949399e9d44/wled00/cfg.cpp#L56-L59), and [2](https://github.com/Aircoookie/WLED/blob/8c44147a4596112a7f5d5720ae809949399e9d44/wled00/wled.cpp#L439-L447) |
| 10 | Clockless LED | Only when creating [`Bus`](https://github.com/Aircoookie/WLED/blob/8c44147a4596112a7f5d5720ae809949399e9d44/wled00/bus_manager.h#L114-L115) |
| many | PWM | Only when creating [`Bus`](https://github.com/Aircoookie/WLED/blob/8c44147a4596112a7f5d5720ae809949399e9d44/wled00/bus_manager.h#L233-L234) |

</details>

Having Tasmota-style templates would be a huge win:
* Better usability for end users ... with friendly names for products (e.g., ESPixelStick v1, v2, v3, etc.)
* Simpler discovery of capabilities for usermod developers (e.g., how many buttons are there?  how many relays? etc.)
* Single location for pin use definition for those bringing up new boards/solutions ... one place to see how all the pins are intended to be used / available for extension

This idea definitely gets my vote!

blazoncek, 2022-10-10:
Implemented in 0.13 as a possibility to apply cfg.json to LED settings page.

closed 2022-10-10 by blazoncek