wled/WLED #1915: Frage zu Playlists

timae146, 2021-04-23:
ich hätte eine frage zu den playlists. 
ich habe mir eine playlist erstellt, die meine Presets 1-An und 2-Aus Ansteuert

`{"playlist":{"ps":[1,2],"dur":[50,10],"end":2}}`

Licht sollte für 5sec an und danach preset 2 für licht aus gestartet.

mein erstes preset sieht so aus:
`{"on":true,"bri":23,"transition":7,"mainseg":0,"seg":[{"id":0,"start":0,"stop":4,"grp":1,"spc":0,"on":true,"bri":255,"col":[[255,10,10],[0,0,0],[0,0,0]],"fx":0,"sx":106,"ix":255,"pal":0,"sel":true,"rev":false,"mi":false},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0}]}`

mein zweites zum ausschalten einfach so;
`{"on":false}`

ich bekomms so aber nicht ans laufen

Aircoookie, 2021-04-26:
Hi!
Die Playlist läuft unendlich durch, immer 5 sec an und 1 sec aus, da kein Wert für `repeat` gesetzt ist.
`{"playlist":{"ps":[1,2],"dur":[50,10],"repeat":1}}` sollte funktionieren :)

timae146, 2021-04-26:
Hi. 
Vielen Dank für die Antwort! Trotz dem, dass du so wenig Zeit für Support hast. Das war der entscheidende Tipp! :)

Mir ist aber noch aufgefallen, das in die playlist noch ein `"on":true` rein muss. Ansonsten gehts auch nicht. Ob das so sein sollte, oder ein bug ist kannst du mir glaube ich am besten sagen :D

Bin davon ausgegangen, das sich das mit den Playlists so wie eine Verknüpfung verhält. Und das `"end":2` die Schleife beendet sobald an ps2 angekommen.

Alles in allem:
`{"on":true,"playlist":{"ps":[1,2],"dur":[30,10],"repeat":1}}`
funktioniert. Danke :>

Aircoookie, 2021-04-26:
Es freut mich, dass es jetzt funktioniert :)
Das mit dem `"on":true` ist in der Tat etwas, woran noch gearbeitet werden muss. Die Playlist soll unterbrochen werden, wenn WLED abgeschaltet wird, das hat aber leider zur Folge, dass das Starten einer Playlist aus dem Aus-Zustand nicht funktioniert.
Benutzt du WLED 0.12 oder 0.11?

Genau so wie eine Verknüpfung funktionieren Playlists auch, nur dass das `end` Preset ignoriert wird wenn `repeat` nicht angegeben ist. Ich werde das Standardverhalten im Fall, dass `end`, aber nicht `repeat` angegeben ist, so ändern, dass `repeat` implizit auf 1 gesetzt wird, das macht das Ganze denke ich etwas intuitiver.

closed 2021-04-26 by Aircoookie

timae146, 2021-04-26:
Hatte es erst mit der 0.12.0 versucht und dann später noch auf die 0.12.1-b1 gegangen, in der Hoffnung das da schon etwas in der Richtung geändert worden ist. 

Okay. Alles klar. Danke für die Erläuterungen.
Und zum Schluss nochmal ein Lob an alle für dieses geile Projekt! :D