wled/WLED #1917: WS2811 12v LED STRIP(3 group of led)  not working with ESP8266

sbali620, 2021-04-25:
[image]

I am using this ws2811 strip with 3 led group.  It run on 12v. now i am using Nodemcu ESP8266 to run this. I am not able to drive it. Now i am not using level shifter. Can it drive without level shifter? I have connected output in D4. So please tell me the solution. how I can use the STRIP? when i connect the power directly to Strip, Its running the effect automatically which is preinstalled.

Aircoookie, 2021-04-25:
Hi!
Can you show us the connections between strip and the ESP8266? WS2811 usually does not have a pre-installed effect, maybe this is a different type of LED chip...

sbali620, 2021-04-25:
Sir, There are 3 pin in led strip. 1-vin 12v, 2- signal ,3-ground. I am using Esp8266 pin D4 connected to signal of Strip. And pin ground is connected to 12v ground and my Ln7805 ground pin. I am using 7805 to powerig esp8266. And I am using it in  Home assistant.

sbali620, 2021-04-29:
> Hi!
> Can you show us the connections between strip and the ESP8266? WS2811 usually does not have a pre-installed effect, maybe this is a different type of LED chip...

[image]
[image]
[image]
[image]

sbali620, 2021-04-29:
i am using even logic shifter

closed 2021-09-07 by stale[bot]