wled/WLED #1918: Brightness fällt automatisch auf 0 ab

Ron0815, 2021-04-25:
WLED wird in Node-Red mit bigtimer zu einem bestimmten Zeitpunkt (Dusk) eingeschaltet. Der gespeicherte Helligkeitswert, den ich im WebUI von WLED in einem Preset hinterlegt habe, fällt immer wieder nach ca. 1 Minute auf 0 zurück. Stelle ich ihn anschließend manuell wieder hoch, bleibt er für nur ca. 1 Minute auf diesem Wert, um dann erneut wieder zu fallen.

Woran kann das liegen?

Viele Grüße, Ron

Ron0815, 2021-04-26:
Sorry for publishing my question/problem in german language. Here' the issue in English:

WLED will be switched on in Node-Red with bigtimer at a certain point in time (dusk). The saved brightness value, which I saved in a preset in the WLED WebUI, always falls back to 0 after approx. 1 minute. If I then turn it up again manually, it stays at this value for only about 1 minute and then falls again.

What can that be?

Best regards, Ron

Aircoookie, 2021-04-26:
Hi Ron!
German is fine for me :) But it will of course limit how others are able to help.

Schick uns doch bitte mal das Preset, welches von Node-Red aufgerufen wird (In der `API command` Box des Presets), ich könnte mir vorstellen, dass die Nightlight-Funktion aktiviert ist, und im Laufe einer Minute wieder auf 0 dimmt.

Ron0815, 2021-04-26:
My preset is configured as follows:

{"on":true,"bri":223,"transition":7,"mainseg":0,"seg":[{"id":0,"start":0,"stop":140,"grp":1,"spc":0,"on":true,"bri":255,"col":[[255,200,147,0],[0,0,0,0],[0,0,0,0]],"fx":0,"sx":128,"ix":128,"pal":0,"sel":true,"rev":false,"mi":false},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0}]}

Ron0815, 2021-04-27:
Hallo,vielen Dank für die schnelle Rückmeldung. Die Einstellung habe ich im issue bei github gepostet in der Annahme, dass das dorthin gehöre. Bin da eber jetzt unsicher und schocke es vorsorglich noch einmal sif diesem Weg:My preset is configured as follows:{"on":true,"bri":223,"transition":7,"mainseg":0,"seg":[{"id":0,"start":0,"stop":140,"grp":1,"spc":0,"on":true,"bri":255,"col":[[255,200,147,0],[0,0,0,0],[0,0,0,0]],"fx":0,"sx":128,"ix":128,"pal":0,"sel":true,"rev":false,"mi":false},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0},{"stop":0}]}Das kann ich alles leider nicht deuten. Siehst Du hier den Fehler, den ich gemacht habe?VG Ron

"target": "https://github.com/Aircoookie/WLED/issues/1918#issuecomment-827008931",

"url": "https://github.com/Aircoookie/WLED/issues/1918#issuecomment-827008931",

"name": "View Issue"

},

"description": "View this Issue on GitHub",

"publisher": {

"name": "GitHub",

"url": "https://github.com"

}

}

]

Ron0815, 2021-04-28:
Liebe Helfer,konnten Sie etwas mit der Rückmeldung anfangen, die ich Ihnen geschickt habe? Ich habe auch noch einmal in der App nachgesehen, ob ich dort irgendwelche Einstellungen falsch vorgenommen habe. Dort konnte ich jedoch eine Einstellung für Nightlight nicht finden. Was ich gefunden habe, war die Einstellung „Timed Light“, wo im Pulldown-Menü die Einstellung „Fade“ eingestellt war. Diese habe ich einmal versuchsweise auf „Wait and Set“ gesetzt. Das half nicht. Im Gegenteil. Nun schaltete sich das Licht nach dem manuellen Ausschalten immer wieder von selber ein. Was kann ich tun?Viele Grüße und vielen Dank für Ihre Mühe,
    	Ron

Ron0815, 2021-04-28:
Additionally, here is the output of a JSON-API query:

{"on":false,"bri":223,"transition":7,"ps":-1,"pl":-1,"ccnf":{"min":1,"max":5,"time":12},"nl":{"on":false,"dur":60,"fade":true,"mode":1,"tbri":0,"rem":-1},"udpn":{"send":false,"recv":true},"lor":0,"mainseg":0,"seg":[{"id":0,"start":0,"stop":140,"len":140,"grp":1,"spc":0,"on":true,"bri":255,"col":[[255,200,147,0],[0,0,0,0],[0,0,0,0]],"fx":0,"sx":128,"ix":128,"pal":0,"sel":true,"rev":false,"mi":false}]}

So, nightlight is disabled ...
Best regards, Ron

Ron0815, 2021-05-04:
Doesn't anyone have any ideas or helpful tips on how to solve the problem? Best regards, Ron

closed 2021-09-11 by stale[bot]