wled/WLED #1920: WLED not discoverd by Alexa

Tedescu74, 2021-04-26:
Hello everyone, first of all a big applause to Aircoookie for this great project! A really cool way to control the LEDs.

About half a year ago I had my WLED ESP8266 connected to Alexa. They were found and everything went great. Now I wanted to control more LEDs and am desperately trying to get them recognized by Alexa (Echo Plus 2nd gen). All settings are correct: 

- Both in the same 2.4 ghz WiFi, 
- WLED unique name assigned, 
- the item "Emulate Alexa Device" is checked etc. 

Have tried it with a NodeMCU V2 and V3, with Wled 0.10, 0.11.1, 0.12 unfortunately no success. 

Looking for solutions, I have also found topics here. The last one with the advice to use the 0.12 version. But unfortunately that did not help either.Then I became aware of the problem with the Hue Bridge V1 in the following thread: 
https://github.com/vintlabs/fauxmoESP/issues/120

Can this be, and can we do something here? The hints with on / off do not work

Aircoookie, 2021-04-26:
Hi, sorry about that!
Alexa support has become very unreliable lately, with various issues popping up.
Here are two things you can try:
- Have you rebooted/power cycled WLED after enabling Alexa emulation in sync settings?
- Have you successfully used the voice control on the ESP8266 that now no longer works before? In that case, Alexa seems to "remember" the old instance (even if manually deleted in the Alexa app!) and refuses to discover the new one...

Tedescu74, 2021-04-28:
Hi Aircookie, after a few days of testing, I got it. I am not sure what was the determining point. I actually think it was an interplay between timing and correct version. And with each new scan, it is mandatory to restart everything e.g Echo, ESP and the Wled app. But anyway, thank you very much for your answer and for your great project.

PalasX, 2021-05-10:
Hey, using release 12.0 on esp8266, can't get Alexa to discover device. Had almost identical issue with a Tasmota install on several devices that was resolved by updating to Tasmota 9.3.1.2. Apparently Amazon altered Alexa's announcement/reply protocol ever so slightly, and using "-1" and the end of the unique device ID wasnt working, where "-01" worked? It's in more detail here:
[https://github.com/arendst/Tasmota/pull/11415](https://github.com/arendst/Tasmota/pull/11415)

In Tasmota, also had to to "SetOption109 1", which forced Alexa Gen1 enabled. Not sure if there is an equivalent in WLED.

Hope this info can help get HUE Emulation / Alexa emulation working again?

dphuk, 2021-06-07:
> Hey, using release 12.0 on esp8266, can't get Alexa to discover device. Had almost identical issue with a Tasmota install on several devices that was resolved by updating to Tasmota 9.3.1.2. Apparently Amazon altered Alexa's announcement/reply protocol ever so slightly, and using "-1" and the end of the unique device ID wasnt working, where "-01" worked? It's in more detail here:
> [arendst/Tasmota#11415](https://github.com/arendst/Tasmota/pull/11415)
> 
> In Tasmota, also had to to "SetOption109 1", which forced Alexa Gen1 enabled. Not sure if there is an equivalent in WLED.
> 
> Hope this info can help get HUE Emulation / Alexa emulation working again?

Same as the above, Tasmota also went a bit nuts recently and any Hue emulated devices dropped from Alexa, including WLED, the Belkin WeMo emulated devices we're all ok though. WLED shows as a hue device and now in the Alexa app just shows as "Device unresponsive". I only really noticed this after the toaster tripper the breaker and all the devices restarted. 

Thing is, it kinda works still, I can ask Alexa to turn on the WLED device, she replies with "Sorry, [Device Name] is not responding.", but the WLED device does turn on though. If I turn it off via Alexa though it seems to turn off and Alexa just bings as if to say it's fine.

AlexKusnezov, 2021-09-30:
@dphuk did you find a solution for that. One of my wled devices becomes unresponsive as soon as I start to control it via the web interface.

closed 2022-04-27 by stale[bot]