wled/WLED #1921: WLED App cant find ESP32

HeathNL, 2021-04-26:
I flashes my ESP32 with the 0.12.0 version after that i reserved a IP adres in my router, put that IP in the WLED  browser wifi config, but when i open the WLED app and press the "+" and try to find my ESP32.... it wont find it.

Aircoookie, 2021-04-26:
Hi!
Since you assigned a static IP for the ESP32, it should be accessible from the phone. You can first try opening the IP in your phone's browser. If it is able to connect, it might indeed be a problem with the app discovery. Are you running the Android or iOS app?

If you can't connect from the browser either, it is a network problem. Either your phone can not reach the network, or the ESP has not properly connected. In that case, check if you can access the IP from a different device. If the ESP failed to connect, it should still broadcast the `WLED-AP` network. You can connect to it and re-enter the WiFi credentials in case they were incorrect. Also make sure the network you are connecting to is 2.4GHz, as 5GHz is not supported. Hope you can get it to work :)

HeathNL, 2021-04-26:
Tnx for the fast reply! I THINK i might have made a mistake,  but correct me if in wrong.

I could not find the ESP32 IP so i made my own static IP in my router and put that IP in wifi settings in the WLAN-AP.

To bad i cant test it right now because the Chinese people from AliExpress try to burn down my house with there bad adapters............ Got to get a new one first -,-"

Aircoookie, 2021-04-26:
Be safe :) You can temporarily connect the ESP32 to any PC via USB (but no LEDs!).

Ah no, that can lead to issues. If the device appears/appeared in your router device menu, you can assign it an IP there, and after a reboot, the ESP32 should acquire that IP automatically from the router.
Alternatively, you can set the static IP in the WLED WiFi settings (and also set the Gateway to the IP of your router).
If you do both, the ESP32 will try to connect with a static IP that is already reserved by the router. It might work, considering it is the same device, but doing one or the other is best. And, if you enter the static IP in WLED settings, setting the gateway is essential!

HeathNL, 2021-04-26:
I could not find it, my router shows only IP adresses and sometimes a device name...

So i hope i can find a step by step tutorial of how to do that, in to noob to setup that kinda things.

If i cant im going to cry xD

Outlook voor Android<https://aka.ms/AAb9ysg> downloaden

huggy-d1, 2021-04-30:
If the router has a DHCP reservations, you can enter the WLED ESP32's MAC address, and then assign an IP address from inside the DHCP address range (some people reserve x.x.x.1 to x.x.x.99 for static IPs, and DHCP starts at x.x.x.100).  

If your DHCP pool starts at x.x.x.100, then a reserved IP should be unused and within x.x.x.100 to x.x.x.254.  

Some people might even restrict it further to something like x.x.x.100 - x.x.x.240 since some devices use the last 16 for special purposes.

If your router has static IP reservations (not DHCP reservations), then they should be outside the DHCP pool range.  Using the above ex: for DHCP pool x.x.x.100 - x.x.x.254, assuming the router is x.x.x.1, then the static IP needs to be unused, not already reserved, and from x.x.x.2 - x.x.x.99.

These things might not apply to your situation.  However, they may help someone else is a similar situation and they find your "issue" and then see my post and find WLED happiness.

ArnabPC, 2021-08-03:
Hello, same issue with me.
My Wifi Router is on Static IP.
I was using ESP8266 and this was my first project, since I'm a dental student and no way relate to this coding and all. I was trying to setup my new room led by sawing a project
Flashing and all done correctly, all LED's are lit. Gave Static IP and Static Default Gateway from my router to WLED app. No device found. Only when connected to WLED-AP, app can discover the lights and can control it. 
Tried via android phone and iPad no device found when on my home wifi.
Please help

Aircoookie, 2021-08-06:
@ArnabPC Hi, you should not enter the Static IP of your router to the WLED WiFi settings.
Instead, enter a static IP within your network that you want the WLED device to have.
The router IP is what you'll usually want to enter for the Gateway IP in WiFi settings.
The easier way is to leave all fields at 0, that way WLED will automatically obtain an IP from your router and you should be able to discover it from the app.

closed 2022-04-16 by Aircoookie