wled/WLED #1923: Which is the best way to create an effect and test it?

Barabba11, 2021-04-26:
Hi! I've looked around an answer for this question, sorry if I just reapeat it another time, 
it's not clear how to create an effect and suddenly test it for debug, I know there are files FX.cpp and FX.h but I guess they need to be compiled and uploaded to the ESP.. looks not that fast.
Or maybe there is another way? A simulator maybe? Thank you a lot!

huggy-d1, 2021-04-30:
If you are building your own WLED using the current master or dev branch, you can use the preview option to have WLED display a visual representation of what the LEDs are doing in the built-in WLED display.  It's one of the top buttons (Peek?).

Barabba11, 2021-04-30:
Thanks for kind answer, maybe I haven't explein it well, I mean how to create a completely new effect and debug it? It seems the only way to it with ESP8266 is to create e new release and upload it, and in this way it's quite impossible to debug.
Maybe there is an emulator or another way to test new effect codes? Thank you

Barabba11, 2021-09-03:
I hope someone may tell me how to develope a new effect (including develop platform/debug) and how to implement it, there is any guide?

blazoncek, 2021-09-03:
> I hope someone may tell me how to develope a new effect (including develop platform/debug) and how to implement it, there is any guide?

Unfortunately only a good sense of logic will help you here.
Keep in mind that effect code is called ~40 times per second (assuming you `return FRAMETIME;`) so serial debugging may not be a good approach.

Look at the existing effects and how they handle segment variables. My recent addition may be useful if you want to learn. Take a look at Tetrix which is generously commented.

closed 2022-04-16 by Aircoookie