wled/WLED #1926: Remember last state before reboot / restart / lost power

Blightbuster, 2021-04-27:
**Is your feature request related to a problem? Please describe.**
I currently have my controller as well as the led strip behind a remote controlled wallplug. I do this so I can fully turn it on and off using google assistant. The problem now is that when I set some preset to be active or just change the brightness, that it will reset to the first preset after every restart instead of resuming with from last state.

**Describe the solution you'd like**
Id would be ideal if the currently applied led config would be stored in persistent memory ( like the general config is ) and gets loaded on boot.

blazoncek, 2021-04-29:
This is available with Auto Save usermod.
Custom build.

amitav2002, 2021-05-17:
How to enable this Auto Save Usermod option.

I am sorry if I am asking that easiest question, but I am new to this.

Please help me out.

blazoncek, 2021-05-17:
> How to enable this Auto Save Usermod option.

You have to custom compile WLED from source and enable usermod by defining #define USERMOD_AUTO_SAVE (or, if using PlatformIO, -D USERMOD_AUTO_SAVE in platformio.ini)
If you are unable to do that you will need to find someone who provides custom binaries or try _bot_ compiles on WLED Discord.

amitav2002, 2021-05-17:
Thanks for the suggestion. I'll try the same.

Blightbuster, 2021-05-17:
Is there a reason for this not being default behaviour?

blazoncek, 2021-05-17:
No, but someone has to do it nevertheless. :)
May be included by default in next release.

Aircoookie, 2021-05-17:
It is not default behavior since you need to write to the (persistant) flash to remember the state between reboots. This could be a problem, since flash has finite write cycles and some cheap ones fail after as little as 1000 writes. Another drawback is that the write operation takes a split second, which can cause the LEDs to stop refreshing for that short time.

There are a few ways to mitigate the impact on flash endurance though:
- Use wear levelling. The LittleFS filesystem WLED is using implements this, so that makes it a lot better already.
- Only save x seconds after the effect was changed. This has the benefit that rapid changes won't cause as many flash writes, but the drawback that if the reboot occurs quickly after setting the desired state, it will not be remembered. This is also used by the usermod.

I am considering adding this functionality natively, as it is requested a lot, but it is not something I would personally use. Of course it increases convenience though :)

blazoncek, 2021-05-17:
@Aircoookie I would recommend to keep it as a usermod (but being compiled in by default).

benji2020, 2021-05-20:
Just have a "save state" button? Gives the user control but reduces wear?

blazoncek, 2021-05-20:
> Just have a "save state" button? Gives the user control but reduces wear?

No. You have Create Preset for that particular request.

sansillusion, 2021-05-20:
You guys are talking about wearing like it would fail after a few saves. I
had a esp32 running a firmware that was saving to spiff every minute or
more for two years before I repurposed it and it still works good. This
only needs to save when there is a change being made. It's not that bad. I
tried the usermod and love it.

blazoncek, 2021-05-20:
Well, as @Aircoookie said, there are flash chips and flash chips.
Generally speaking they have several 100.000's writes which would allow one to save almost every second and still use a device for a year without issues. But then there are some that die just after a few 100's of writes. YMMV.
Auto save usermod has a built-in delay of (configurable) 15s so in the worst case scenario it will only write 5.760 times a day. Much less in a real world situation. If you are constantly playing effects on a loop, you may better disable usermod nevertheless.

Aircoookie, 2021-05-23:
In addition to the flash wear, I need to share that I am also a little worried about #1981 , corrupted `presets.json` files come up from time to time. I don't know whether the problem lies in my file manipulation code or in the LittleFS handling of partially rewriting large files (it's bad, if you want to replace something in the beginning of the file, it will need to rewrite everything in the file after the location you want to change), but it is worrying.

Aircoookie, 2021-05-23:
Maybe that risk could be mitigated by using a second `asave.json` file that only contains the single auto-saved preset.

blazoncek, 2021-05-23:
Indeed I have experienced a cfg.json corruption a few days ago on an ESP01 with no writes to flash running for 80 days without interruptions.
I would assume it is a flash fault and not LittleFS.

blazoncek, 2021-10-03:
Recent code has been optimized to reduce flash writes even further keeping your flash memory healthy even when using Auto Save usermod and changing presets very often.
Still recommending to include Auto Save usermod in default builds.
Closing.

closed 2021-10-03 by blazoncek

asking23, 2022-04-19:
I have enabled -D USERMOD_AUTO_SAVE feature. But its not working. Please help what could be wrong ?

[image]

blazoncek, 2022-04-19:
Check if you also have boot preset assigned in LED preferences.