wled/WLED #1927: Lava Lamp Effect Idea

Doyle4, 2021-04-27:
I was watching a youtube video [https://www.youtube.com/watch?v=C4QELMzOpiA](url) and spotted an effect reminded me of a Lava Lamp.

At 21:06 is the effect.

Could this be tweaked, say the white in that effect was white wax that moves slowly down with oddments moving up, small ones drop down and a large wax bit moves up. When wax is at the bottom make the white leds glimmer to give effect of the wax moving around at the bottom.

Using the 3 colour selections, 1 colour could change wax colour, 2nd one water and 3rd lava lamps bulb.

Just an idea :)

Aircoookie, 2021-05-17:
Hi!
The effect in the video looks like a standard Rainbow effect, which you can get in WLED by using the `Palette` effect with the `Rainbow` color palette.
For an effect that look more lava lamp like, the Noise effects with really slow brightness and appropiate palette may also be nice.

That said, a dedicated, smooth, slow lava lamp effect would be great to have and I will think about how it could be implemented!

hell1337hound, 2021-05-22:
+1 thanks Aircookie!
Heres one i made for my bathroom that acts quite a bit like a lava lamp
https://pastebin.com/uL6V6XwD