wled/WLED #1928: PIR_sensor_switch

ItsMrJP, 2021-04-27:
I am trying to reduce the user mod for the PIR sensor switch so that the light only turns on for 15-30 seconds. There is a limit in the range. Currently the lowest possible value is 60 seconds. Any help would be much appreciated!

blazoncek, 2021-04-29:
I am currently working on a fully configurable PIR sensor switch usermod.
It is already available in my WLED repo, dev branch.
It may take a while to get included into WLED master though.

1igeza, 2021-04-30:
Will it be possible to use more then one PIR sensor and send also status to mqtt?

blazoncek, 2021-04-30:
More than one, no.
MQTT, yes.

ItsMrJP, 2021-05-04:
> I am currently working on a fully configurable PIR sensor switch usermod.
> It is already available in my WLED repo, dev branch.
> It may take a while to get included into WLED master though.

Awesome! Thank you! Will check out your fork

blazoncek, 2021-05-07:
This has been now submitted as a PR for master.

closed 2021-09-07 by blazoncek