wled/WLED #1933: Random effect via JSON API

MacGyverBeta, 2021-04-28:
Hello together,

it would be awesome to have the possibility to set random effects via the JSON API.

I would like to trigger a random effect(and maybe color) via a button press and the only solution I see is to set a preset via the button.
But the presets use the JSON API syntax and this does not have a support to cycle through the effects, 
like the [HTTP request API - In-/decrementing values ](https://github.com/Aircoookie/WLED/wiki/HTTP-request-API#in-decrementing-values) does have.

Thank you for your awesome software!

Legsmaniac, 2021-04-29:
I imagine it would be easy enough to create a Custom Usermod to create such effect.
See here - https://github.com/Aircoookie/WLED/wiki/Add-own-functionality

Where btnON would be an available pin on your Arduino, the basic code in its purest form could be something like....

```
void loop() {
if (digitalRead(btnOn) == HIGH)    {
    //  Get a random FX
    randomSeed(analogRead(0));
    randomFX = random(0, MODE_COUNT-1);
}
  //  Set strip to random FX
  strip.setMode(strip.getMainSegmentId(), randomFX);
}
```
Then with each button press, a random effect would be generated and selected.
Of course, this basic code only allows for the whole main segment so assumes you do not have segments set.

Aircoookie, 2021-05-01:
Hi, you are absolutely correct, it would be awesome to have support both for setting random values and incrementing/decrementing via JSON!
I'll think about how to best implement it as we cannot prefix `~` for in/decrementing like in HTTP API.
What would be possible is accepting strings in addition to numbers for fields like brightness/fx/speed/intensity/preset, where `"r"` could indicate a random value and `"+x"`,`"-x"` for in/decrementing the current value by x.

MacGyverBeta, 2021-05-02:
> Hi, you are absolutely correct, it would be awesome to have support both for setting random values and incrementing/decrementing via JSON!
> I'll think about how to best implement it as we cannot [...]

Sounds like an even more powerful solution! Awesome idea!

blazoncek, 2022-07-25:
Implemented in 0.13

closed 2022-07-25 by blazoncek