wled/WLED #1934: How to allocate GPIO during compilation and fix it to prevent changes in UI interface?

zx804262156, 2021-04-30:
How to allocate GPIO during compilation and fix it to prevent changes in UI interface?

Aircoookie, 2021-04-30:
Hi, you can pre-set GPIOs in your platformio environment with defines:
To init 4 busses with 30 LEDs each:
```
-D PIXEL_COUNTS="30,30,30,30" -D DATA_PINS="16,1,3,4" -D DEFAULT_LED_COUNT=120
```

It is also possible to pre-set the pin for the button, IR, and relay functions, e.g.:
```
-D IRPIN=5 -D BTNPIN=4 -D RLYPIN=13
```

Preventing pin changes in UI is not something I am willing to support (though if you only need one output, you could use WLED 0.11.1, which does not support in-UI pin selection). In the next major version I will add a kind of simplified mode, that may hide pin choices unless the user switches into advanced settings mode.

zx804262156, 2021-05-06:
I found that if the wrong button gpio pin is changed, the button cannot be reset when the wifi cannot be connected. So I want to set the button GPIO pin at compile time. And prevent modification on the WEB interface.

blazoncek, 2021-05-17:
You cannot prevent modifications except if you password protect Settings page.

tarontop, 2021-08-10:
Hello everyone, I want to call the reset function to reset wled, which one is the reset function? Because I need a reset in my project

tarontop, 2021-08-10:
Hello everyone, I have a RGBWC bulb, how can I preset GPIO before the compiler and set the default module to RGBWC instead of WS281x

closed 2022-04-16 by Aircoookie