wled/WLED #1935: Switched off sucessfully, but after time one led appered ON

Barabba11, 2021-04-30:
I tried now switch on and off, sometimes, randomly, it still a led switched on (ony one on 144,  always in the end of stripe, maybe 143, 141.. etc, I'm using version 0.12) so I'm not sure it appeared after time of just after switch off I haven't noticed it (led was really dimmed to minumum brightness, according to the effect I was running).
It's impossible to determine the effect running after restart the app on my phone, so I can't teel you exactly the name of effect, sorry

Aircoookie, 2021-04-30:
Hi!
This is most likely a data signal integrity issue, often caused by RF interference near the affected LEDs.
WLED stops refreshing the LEDs when turning off, therefore such glitches will cause the LED to stay lit.
It might also help to use a levelshifter if you are not yet (I recommend the SN74AHCT125N)

Barabba11, 2021-04-30:
Level shifter is really a good idea! I haven't considered ther ESP8266 has only 3,3v vs 5V led. Anyway I believe the problem appears dung the fade off of the swithching off procedure, Appears after 0,5 second. It may be as you said determined by a lack of voltage, but I've never noticed any mistakes when switched on.
In my opinion is a combnation between that effect + switch off, you may introduce a double reset to zero to all led to guarantee more the off status. Thank you :)

Barabba11, 2021-08-30:
You may decide to close this notification, I may say that this issue can be related to a (or just some) specific effects only. When I'll catch which one I'll write here

closed 2022-04-16 by Aircoookie