wled/WLED #1937: Button doesn't work version 0.12.0

DeamBeam, 2021-05-01:
Button doesn't work on 0.12.0. It does nothing when pressing the button.
Flashed an old Firmware i had lying around (version 0.10.0) and it worked fine there. Then tried the beta version 0.12.1-b1 and the button also works fine there.
Using an NodeMCU with the standart ESP8266 bin.

Also a quick question: How do you cycle through the presets with the new macro? I want to cycle through presets with a double press on the button.

blazoncek, 2021-05-03:
You have to configure button GPIO pin in LED settings page.

Aircoookie, 2021-05-03:
And don't forget to reboot after setting the pin! (set pin 0 for D3)

Diranar, 2021-05-04:
For me it does not work either. Ive used Button pin 0, this is also changed in the config. if i Press the Button is see that it works, cause a LED on the ESP reacts to the Button presses. But i want it the old way that a shrot Press changes Presets, Long Press for On/OFF. How does this work now ? Can you write a little Tutorial cause it makes no sense now.

...And please don't say i have to adjust every single Preset i made, that would be super user unfriendly and a real Reason to turn back to WLED 0.0.10 or before on all my esps.

This new Macro usage in the Jason is weird and was way better in versions 0.10 and before, cause IT WORKED ! Now its just weird with all this Programming stuff, not User firendly anymore i have to say, if you don't hava an informatic diploma. I want that it just works and is as easy to set up as the old Version.

blazoncek, 2021-05-04:
<sarcasm on>
Why did you upgrade then, if old was so great?
Revert back and that's it.
You want new features? Well they come at a price.
<sarcasm off>

Unfortunately it is not possible to satisfy _all_ users. If you need new features you will need to adapt to _all_ of them, including presets. They offer a ton of new possibilities, including playlists, at the cost of learning a few new things (which frankly aren't that complicated and you have examples in the wiki).

If you don't want to learn new ways you can stay with the version you like most. Nobody is forcing you to upgrade. And if your ESPs are working great and fulfill all your desires there is really no need to upgrade.

Diranar, 2021-05-04:
>  Why did you upgrade then, if old was so great? Revert back and that's it. You want new features? Well they come at a price.
> 
> Unfortunately it is not possible to satisfy _all_ users. If you need new features you will need to adapt to _all_ of them, including presets. They offer a ton of new possibilities, including playlists, at the cost of learning a few new things (which frankly aren't that complicated and you have examples in the wiki).
> 
> If you don't want to learn new ways you can stay with the version you like most. Nobody is forcing you to upgrade. And if your ESPs are working great and fulfill all your desires there is really no need to upgrade.

Thanks for your useless opinion, that doesn 't help at all and is a waste of my Time, so i like to ignore this and hope a more helpfull user can write a reasonable "How to".

Legsmaniac, 2021-05-05:
> ... and hope a more helpfull user can write a reasonable "How to".

**And along I pop with a more reasonable and helpful How To for you.**  😊

Step 1 - On all your ESPs running the WLED Version 0.11.0 and above which for you _"This new Macro usage in the Jason is weird and was way better in versions 0.10 and before, cause IT WORKED"_, wipe the ESP to entirely remove and reset all traces of code and memory storage.

Step 2 - Download the [WLED Version 0.10.2](https://github.com/Aircoookie/WLED/releases/tag/v0.10.2) which avoids all versions above WLED 0.11.0 because _"its just weird with all this Programming stuff, not User firendly anymore"_

Step 3 - Now carefully flash all your newly wiped ESPs with the downloaded WLED version 0.10.2 because you _"want that it just works and is as easy to set up"_

Step 4 -  Sit back and enjoy your now super-friendly ESPs running WLED version 0.10.2 because you find them _"a real Reason to turn back to WLED 0.0.10 or before on all my esps"_ - it really is that easy and simple!  😉

Diranar, 2021-05-05:
Thanks, thats funny and it doesn't help cause this would wipe all my patterns and Presets, so your tutorial is of no use besides educational purpose, which is very low, too.

If a user could give a Tutorial how to use the Button with the new 0.12 Version. That would be helpfull.
Until that i presume the whole Button Function is broken and does not work at all. 
- Like the worthless Tips of the not helpfull "Community" here.

blazoncek, 2021-05-05:
How much did you pay for WLED? Nothing?
If the button is broken for you, fix it. Nobody is stopping you. You may even get credit as a contributor.
Stop whining and start experimenting and read the wiki. All the information needed to make button _work_ is there.

Diranar, 2021-05-05:
> 
> 
> How much did you pay for WLED? Nothing?
> If the button is broken for you, fix it. Nobody is stopping you. You may even get credit as a contributor.
> Stop whining and start experimenting and read the wiki. All the information needed to make button _work_ is there.

Stop making Senseless comments and try to be a good Contributor and help the Community instead of just giving useless tips would be a great start for you ! 
Stop Wining yourself first, this is getting boring with small Minded People like yourself. XD

blazoncek, 2021-05-05:
> with small Minded People like yourself. XD

There is no need to be offensive.

Aircoookie, 2021-05-05:
Woah, maybe I should have merged the toxicity detection bot after all 😄 @Diranar that's not the kind of language I want to see here. I know how frustrating it is if something doesn't work, but we are not getting paid for helping you, this whole project and support is all done in our free time.

Back on topic, button in 0.12 is still easy:
- Set a button pin in LED settings (e.g. 0), save, and reboot WLED. (this will be set as default in the future again, but a bug caused new 0.12 installations to disable the button by default)
- Button will now use the default behavior (short press on/off, long press change to random color)
- Save a preset for what you want the button to do. Can be off, an effect, whatever.
- Take note of the preset ID (number to the left of the preset name)
- Put that number into the short/long/double press macro field and save time settings.
- If you really want to use HTTP macros instead of JSON or complete presets, that's of course still possible. Just uncheck _Use current state_ and enter your macro (e.g. `A=255`) into the "API command" field. I'd argue that this new system (which exists since 0.11) is more "user friendly" than the old one, since it merges the functionality of macros and presets into a single, powerful feature

closed 2021-05-05 by Aircoookie

Diranar, 2021-05-06:
Thank you. 
Maybe you should update the Dokumentation and Wiki after an Sofware update, too, so that Future Questions regarding a Button are not needed, cause then its dokumented how it works.