wled/WLED #1939: New api feature request:  gauges

trv-in-3d, 2021-05-02:
**Describe the solution you'd like**

I  would like to be able to configure a segment and tag it as a "gauge". 
Doing so would require: 
-       (obviously the pixel range)
-      "low" value - default 0
-      "high" value - default 100
-      "Gauge type" -  default "bar"
-      Possibly a "Gauge number" - this would allow the section to be referenced by a new API call:  SetGaugeValue

Now an api call to "SetGaugeValue" would pass {GaugeID:1, Value:75}
In this case, 3/4 of the LEDS would be lit.

This capability would allow me to make what are basically static calls from other environment and basically pass a single "value" to see progress.

I can also see some other uses:   setting the "low" and "high" values from -10 to 110 may let me plot outdoor temperatures.

Setting the "low value" at 150 and the "high value" at 260 would let me plot hot end temps from a 3d printer.

I also am picturing a few different "Gauge types"
- "bar" would light from the low pixel to the high based on the value - as described above.
- "point"  would light a single pixel in the range based on the value
- "brightness" would light the whole gauge range - with the "value" driving the intensity.
- "pallet"  would light the whole gauge range using the first color in a defined pallet as the "low" value and the last one as the "high" value

 In my ideal world, all these could be combined.   So a "point","pallet" gauge may show LED 0 as the first LED dimmly lit in Red for a value of 0 and value of 100 as LED 8 brightly lit white if the "candle" pallet was also refenced to that section. 

When all this is said and done, I can use an 8x8 led array to become an indicator of 8 different values at a glance.  The heavy lifting of "Creating the visual" is done by wled and all the remote systems have to do is pass in a single value.

sansillusion, 2021-05-02:
That sounds like a great idea and lots of use possibility.

sansillusion, 2021-05-02:
I would love to have mode "bar" and "palette" combined so that my gauge could be changing color as well as nuber of leds that light's up. That eay I could kind of match them up with some gauges in my Home Assistant.

Aircoookie, 2021-05-03:
Hi! This functionality is possible using the "Percent" effect!
By setting its intensity value in the range 0-100, that part of the segment will be lit.
The WLED plugin for Octoprint uses this too for hotend temps and print progress.
The remote software will need to do the mapping from the input value range to 0-100, but otherwise it should cover your usecase pretty well.
"Bar" and "Palette" gauge types are currently implemented by using different colors and palettes in WLED, a "Point" mode would be a nice addition indeed.

sansillusion, 2021-05-03:
I did not think of that, great idea I can send a mapped value to in persent effect and set color trough HA. Great I now have a new project ! I want to map all my grow room sensor gauges to wled gauges !

blazoncek, 2023-05-27:
Closing as it is implemented as an effect as described above.

closed 2023-05-27 by blazoncek