wled/WLED #1940: Support for esp32-c3

danielclasen, 2021-05-03:
**Is your feature request related to a problem? Please describe.**
Currently flashing the binary onto a board featuring the esp32-c3 like the [ESP32-C3-DevKitM-1](https://docs.espressif.com/projects/esp-idf/en/latest/esp32c3/hw-reference/esp32c3/user-guide-devkitm-1.html) does not work. The serial console just spams _Invalid Header_ and the board is constantly rebooting.

**Describe the solution you'd like**
Would it be possible to add support for this kind of board/mcu? Apparently there is an early stage of the arduino-espressif framework that might already be usable in platformio as it is being discussed here: https://github.com/platformio/platform-espressif32/issues/497

**Describe alternatives you've considered**
-/-

**Additional context**
I am sadly quite unfamiliar with that kind of hardware development but if I can help with anything or test something on one of my boards, don't hesitate to ask! I also have a few spare ones, so If you'd really like to dig down that issue I can send you one of those ESP32-C3-DevKitM-1 for debugging/testing/development purposes.

Thank you for your hard work and effort to make WLED even better! :)

Aircoookie, 2021-05-03:
Hi!
I can confirm that ESP32-C3 is not currently supported. We'll need to get decent Arduino Core support for it first, which usually takes a while. We are currently looking to get ESP32-S2 supported, and that has been out since 2019 :)
Definitely will support the C3 though once the core becomes stable/usable.

danielclasen, 2021-05-03:
Hey @Aircoookie

thx for letting me know :) Give me a heads up once the arduino core lib is ready and you start integrating it. I would really like to contribute, even if that means providing you and other maintainers with some test boards is all I can do :D

Cheers,
Daniel

frederictobiasc, 2021-06-25:
Hi @Aircoookie,
the issue @danielclasen linked was closed just a few days ago. Does this mean, supporting esp32-c3 could move forward?

tony-fav, 2021-09-17:
Definitely tracking this issue now myself, starting to use more C3s in projects personally than 8266s.

hutje, 2021-12-26:
Support for Esp32-C3 would be really useful.

misery, 2021-12-28:
Maybe someone wants to try it: https://github.com/Aircoookie/WLED/pull/2454

```pio run -e esp32c3```
I can link a binary now. But I don't have a C3 yet.

You should expect problems as it grabs a lot of "current development" branches.

oldcrazyeye, 2022-03-09:
I'm wondering the same! Have a bunch of c3's collecting dust over here

misery, 2022-03-10:
Well, it was merged into master branch. So you can start using it. :-)

peturdainn, 2022-04-03:
I just tried WLED on an ESP32c3 (build from master branch), and I have 2 issues:

1) the board I use (https://www.banggood.com/ESP32-C3-Development-Board-RISC-V-WiFi-Bluetooth-IoT-Development-Board-Compatible-with-Python-p-1914005.html) has 25 LEDs on GPIO 8, which I can use with the neopixel lib in Arduino, but WLED does not allow GPIO8. 
I changed the code that prevents using GPIO8, so config accepts 8 as LED GPIO but the LEDs do not work.

2) As soon as I enter wifi settings, WLED seems to reboot at a regular interval (I see the AP come up and go away again), I do not see it in the client list of my AP.

Are any of these 2 known issues? Can I help?

misery, 2022-04-05:
>     2. As soon as I enter wifi settings, WLED seems to reboot at a regular interval (I see the AP come up and go away again), I do not see it in the client list of my AP.

Seems like this: https://github.com/Aircoookie/WLED/issues/2518

Jugrnot, 2022-04-07:
> Well, it was merged into master branch. So you can start using it. :-)

Any tips on how I can get wled to actually flash and boot on an esp32c3 based board?  I bought a boat load of boards for a ton of wled projects, only to discover that apparently wled just flat out doesn't work on these???  Based off the wiki, I ordered some of [these ](https://www.electrodragon.com/product/esp-led-strip-board/) from electrodragon, and I'm something like 7 hours into failing at successfully flashing even one of them.  Absolutely every single thing I do, up to and including downloading the source and compiling specifically the esp32c3 bin, on boot the serial output is just "invalid header."

All flashing has been attempted with espeasy and esphome flasher.  I've also attempted flashing every bootloader under the sun I can find for an esp32c3 and only results in the exact same thing.

Specifically I've tried flashing a bootloader aircookie suggested in another github thread I ran across, and I'd link that here, but I've been to a hundred different github repos today trying to find an answer for this and can't readily find it..  Appreciate any insight anyone could give on this!

blazoncek, 2022-04-08:
Try using official Espressif flashing tool ‘esptool‘.

Jugrnot, 2022-04-08:
> esptool

Biscuits..  Neglected to mention, I've tried esptool as well by flashing various bootloaders to 0x0 and the wled bins to 0x10000.  Have also used esptool to erase the flash and start over as well.

> esptool.py write_flash 0x0 "esp_bootloader_v4.bin" completes in ~2 seconds
> esptool.py write_flash 0x10000 "wled_013.1_esp32.bin" completes in ~60 seconds

Enable a serial monitor, reboot esp and just dumps the invalid header.

Just gave the expressif 'esptool' a shot, flashing the bootloader to 0x0, rebooting into flash then the wled bin to 0x10000 with the same result.  Initial serial connection gives a bunch of garbage ascii characters, then floods with invalid header again.

peturdainn, 2022-04-08:
@Jugrnot there is no official binary yet, so the one you used, if downloaded from here, isn't for the c3. The c3 has another processor core than the other ESP32 family so other instruction set.
Only master branch has support so you have to build yourself.
And I think there are some rough edges still with regard to c3 support so give it some time & relax.

PW999, 2022-07-03:
> Biscuits.. Neglected to mention, I've tried esptool as well by flashing various bootloaders to 0x0 and the wled bins to 0x10000. Have also used esptool to erase the flash and start over as well. [...]

You need another bootloader for the ESP32-C3, you can grab them here: https://github.com/espressif/arduino-esp32/tree/master/tools/sdk/esp32c3/bin

With the help of this comment I managed to get a version that at least boots :)

Jugrnot, 2022-07-21:
> 

Well I finally got around to giving this another go with your suggestions, but quite honestly my level of knowledge with programming just isn't up to snuff with this stuff and frankly I don't have the time to mess with this anymore.  Bought these because they were (I thought) on the wiki list of compatible devices, not realizing that it was the esp8266 versions specified and not the esp32.  Think these will end up on ebay or something and I'll just stick with what I know and what works.

Thanks anyway!

PW999, 2022-08-14:
I managed to get WLED working on the ESP32-C3, it came down to the changes from @Depau which never got merged into ESPAsyncWebserver: https://github.com/Depau/ESPAsyncWebServer/tree/wi-se-patches (link to the original issue: https://github.com/me-no-dev/ESPAsyncWebServer/issues/951)

For testing purposes I updated ESPAsyncWebserver to `https://github.com/Aircoookie/ESPAsyncWebServer.git @ ~2.0.5` and then patched the code locally. I still need to cleanup my config, but it's currently this

```
[env:esp32c3]
board = esp32-c3-devkitm-1
;platform = espressif32@5.1.0
platform = https://github.com/tasmota/platform-espressif32/releases/download/v2.0.2.2/platform-tasmota-espressif32-2.0.2.zip
platform_packages =

;platform_packages = framework-arduinoespressif32 @ https://github.com/Aircoookie/arduino-esp32.git
framework = arduino
board_build.partitions = tools/WLED_ESP32_4MB_1MB_FS.csv
upload_speed = 115200
build_unflags = ${common.build_unflags}
lib_deps = ${esp32c3.lib_deps}
build_flags = -g
  -DARDUINO_ARCH_ESP32
  -DARDUINO_ARCH_ESP32C3
  -DCONFIG_IDF_TARGET_ESP32C3
  -D CONFIG_ASYNC_TCP_USE_WDT=0
  -DCO
  ;-DWLED_DEBUG_FS
; ; ******** non-standard build flags defined below
  ;${common.debug_flags}
  ;-DDEBUG_ESP_CORE
   ;-D WLED_USE_MY_CONFIG
   ;-D USE_APA102
   ;-D CLKPIN=6
   -D DATAPIN=5
   -D SERVERNAME="\"WLED\""
   -D DEFAULT_LED_COUNT=30
   -D ABL_MILLIAMPS_DEFAULT=2500
```

PW999, 2022-08-15:
It still crashes from time to time, but there is no specific action that triggers this, it's just a matter of randomly hitting stuff in the UI. Log output is from a build with `${common.debug_flags} -DDEBUG_ESP_CORE`
```
Terminal log file
Date: 15/08/2022 - 09:06:05
-----------------------------------------------
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DOUT, clock div:2
load:0x3fcd6100,len:0x38c
load:0x403ce000,len:0x6ac
load:0x403d0000,len:0x24e4
SHA-256 comparison failed:
Calculated: aef91b0b298f42b97535c16d8c14ff07892a660497e887393451589b96b8edda
Expected: ffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffffff
Attempting to boot anyway...
entry 0x403ce000

---WLED 0.13.2-a0 2203191 INIT---
esp32 v4.4-3-g6afb23d90a
heap 249100
Watchdog enabled: OK
PIN ALLOC: Pin 1 allocated by 0x89 (137)
Registering usermods ...
Reading config
Reading settings from /wsec.json...
JSON buffer released. (3)
Reading settings from /cfg.json...
Removing all.
PIN ALLOC: Pin 5 allocated by 0x82 (130)
Successfully inited strip 0 (len 68) with type 22 and pins 5,255 (itype 17)PIN ALLOC: Pin 0 allocated by 0x85 (133)
Starting usermod config.
JSON buffer released. (1)
Initializing strip
Usermods setup
No connection configured. Opening access point WLED-AP
Init AP interfaces
Connected AP clients: 1
Not-Found HTTP call:
URI: /generate_204
FileRead: /generate_204
FileRead: /index.htm
Not-Found HTTP call:
URI: /gen_204
FileRead: /gen_204
JSON buffer size: 1729
JSON buffer released. (17)
Not-Found HTTP call:
URI: /presets.json
FileRead: /presets.json
JSON buffer released. (12)
Not-Found HTTP call:
URI: /generate_204
FileRead: /generate_204
Not-Found HTTP call:
URI: /generate_204
FileRead: /generate_204
FileRead: /index.htm
Not-Found HTTP call:
URI: /generate_204
FileRead: /generate_204
Not-Found HTTP call:
URI: /generate_204
FileRead: /generate_204
Not-Found HTTP call:
URI: /skin.css
FileRead: /skin.css
JSON buffer size: 3972
JSON buffer released. (17)
[... 116 lines cut ...]
esp32 v4.4-3-g6afb23d90a
heap 249100
Watchdog enabled: OK
PIN ALLOC: Pin 1 allocated by 0x89 (137)
Registering usermods ...
Reading config
Reading settings from /wsec.json...
Removing all.
PIN ALLOC: Pin 5 allocated by 0x82 (130)
Successfully inited strip 0 (len 68) with type 22 and pins 5,255 (itype 17)PIN ALLOC: Pin 0 allocated by 0x85 (133)
Starting usermod config.
JSON buffer released. (1)
Initializing strip
Usermods setup
JSON buffer rializing strip
Usermods setup
rom /cfg.json...
No connection configured. Opening access point WLED-AP
Init AP interfaces

-----------------------------------------------
Date: 15/08/2022 - 09:07:11
End log file

```

blazoncek, 2022-09-15:
Anyone interested may try https://github.com/blazoncek/WLED/tree/esp32-s2 
Please post your working PIO environment for others.

AllExHH, 2022-09-29:
Hey! Any news about esp32 c3?

blazoncek, 2022-09-29:
Just got them today.

AllExHH, 2022-10-02:
I have try it at Seeed Studio Xiao esp32c3, it's runing!

blazoncek, 2022-10-03:
Looks like:
```
platform = espressif32@5.1.1
framework = arduino
board = esp32-c3-devkitm-1
;board_build.flash_mode = qio
;board_build.f_flash = 80000000L
board_build.partitions = tools/WLED_ESP32_4MB_1MB_FS.csv
upload_port = /dev/cu.usbmodem1A12301
monitor_port = /dev/cu.usbmodem1A12301
build_flags = ${common.build_flags} ${esp32c3.build_flags} #-D WLED_RELEASE_NAME=ESP32-C3
  -D WLED_WATCHDOG_TIMEOUT=0
  -D LEDPIN=8
```
produces working environment.

blazoncek, 2022-10-05:
Looks like the platform espressif32@5.1.1 and above environment produce a working ESP32-C3 environment.
Closing.

closed 2022-10-05 by blazoncek

Aircoookie, 2022-10-07:
Confirming it seems to work great so far on the tiny Seeedstudio Xiao!

Note that might be helpful to someone "bricking" the board (software USB inaccessible) with the following error: 
```
esptool failed to connect: Cannot configure port, something went wrong. Original message: PermissionError(13, 'A device attached to the system is not functioning.', None, 31)
```

-> Hold flash button, press reset button, release flash button again. C3 will enter download mode and esptool works again.
After erasing/uploading, press reset again to boot uploaded code.

blazoncek, 2022-10-07:
My various test (and I did plenty) showed you cannot "brick" ESP as long as you have access to strapping pins (GPIO0 on ESP8266, ESP32 & ESP32-S2, GPIO9 on ESP32-C3) as you can always put ESP into "firmware upload" mode (also called DFU sometimes).
The problem with ESP32 variants is that they also need *correct* flash content in the first 64kB (bartitions, OTA config, etc) also called "bootloader". This "bootloader" mostly depend on the flash size used.

magiclightde, 2022-10-11:
How can i use the GPIO8 RGB LED on the ESP32-C3-DevKitM-1?

a-a-crabtree, 2022-11-04:
> I have try it at Seeed Studio Xiao esp32c3, it's runing!

@AllExHH Can you share your PIO & bootloader setup? I've tried with the one here for the XIAO and haven't been able to get it out of the bootloop.

blazoncek, 2022-11-04:
Please join Discord or WLED [forum](https://wled.discourse.group) for help and support questions.

PeterGrace, 2022-12-15:
For those having issues with esp32-c3 on main branch, as of today, 12/14/2022, I had to upgrade espressif@5.1.1 to espressif@5.2.0 in the platformio.ini.  Once that was done, my device (an ESP32-C3-DevKitC-02) went from bootloop with an lfs error, to functioning appropriately.
```
[env:esp32c3]
platform = espressif32@5.2.0
framework = arduino
board = esp32-c3-devkitm-1
board_build.partitions = tools/WLED_ESP32_4MB_1MB_FS.csv
build_flags = ${common.build_flags} ${esp32c3.build_flags} #-D WLED_RELEASE_NAME=ESP32-C3
  -D WLED_WATCHDOG_TIMEOUT=0
upload_speed = 460800
build_unflags = ${common.build_unflags}
lib_deps = ${esp32c3.lib_deps}
```

limengdu, 2023-01-16:
> 

Can you tell me how to configure it so that WLED can run smoothly on XIAO ESP32C3? I'm really interested in this! Thank you for your help.

blazoncek, 2023-01-16:
@limengdu Please join Discord or WLED [forum](https://wled.discourse.group/) for help and support questions.

wilson3682, 2023-01-19:
@limengdu Check this link. I created a platformio_override.ini with the info provided and was able to upload wled to XIAO ESP32C3.
https://github.com/Aircoookie/WLED/issues/1940#issuecomment-1214424062

ozett, 2023-01-19:
would love to follow a solution here in this thread...
maybe thats possible?

wilson3682, 2023-01-20:
@limengdu This is what I did and it is working so far with xiao esp32c3. 

I followed 
@PW999 https://github.com/Aircoookie/WLED/issues/1940#issuecomment-1352391486
@PeterGrace https://github.com/Aircoookie/WLED/issues/1940#issuecomment-1352391486
and @Aircoookie https://github.com/Aircoookie/WLED/issues/1940#issuecomment-1271029990
recommendations an created a platformio_override.ini with the following code:

[platformio]
default_envs = esp32c3

[env:esp32c3]
board = esp32-c3-devkitm-1
;platform = espressif32@5.1.0
platform = espressif32@5.2.0
platform_packages =
framework = arduino
board_build.partitions = tools/WLED_ESP32_4MB_1MB_FS.csv
upload_speed = 115200
build_unflags = ${common.build_unflags}
lib_deps = ${esp32c3.lib_deps}
build_flags = -g
  -DARDUINO_ARCH_ESP32
  -DARDUINO_ARCH_ESP32C3
  -DCONFIG_IDF_TARGET_ESP32C3 -W
  -D CONFIG_ASYNC_TCP_USE_WDT=0
  -DCO
  -D WLED_DISABLE_ALEXA
  -D WLED_DISABLE_BLYNK  
  -D WLED_DISABLE_INFRARED
  -D DATAPIN=9
  -D SERVERNAME="\"WLED\""
  -D DEFAULT_LED_COUNT=30

Hope it works for you.

PW999, 2023-01-23:
I tried @wilson3682 's above config on two ESP32-C3-DevkitC-02 v1.1's and both seem to be working fine. Even from a completely wiped flash this works fine.

The onboard RGB LED on GPIO8 can also be controlled from WLED.

softhack007, 2023-01-23:
> build_flags = -g
>   -DARDUINO_ARCH_ESP32

You need to include `build_flags = ${common.build_flags} ${esp32c3.build_flags}`, otherwise you get all sorts of crazy behaviour later, because you are missing global options.

Also I would not overwrite an existing build target, but rename your build target to something like  `env:my_esp32c3dev`.

lsiudut, 2023-01-25:
I didn't want to post here after using forum was suggested, but I also don't have slightest problems with S3 for over a year atm. I do normal platformio build for the s3 platform (with few minor code modifications to allow using gpios > 33).

softhack007, 2023-01-25:
> I also don't have slightest problems with S3 for over a year atm

Lucky you :-) great that it works for you, as owners of other -C3/-S2/-S3 boards still experience a lot of problems. 

> with few minor code modifications to allow using gpios > 33

These modifications are not needed any more.

Actually the best approach - as you suggest - is to build yourself in platformio. Pre-build .bin files are not working everywhere.
btw, make sure you __dont__ flash the standard ESP32 bootloader from the WLED release page.

WAPEETY, 2023-03-05:
Soo, I'm following this and so far I have  this output log... any idea?

```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0xf (BROWNOUT_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DOUT, clock div:1
load:0x3fcd6100,len:0x38c
load:0x403ce000,len:0x6ac
load:0x403d0000,len:0x24e4
entry 0x403ce000
[   333][E][vfs_api.cpp:104] open(): /littlefs/ledmap1.json does not exist, no permits for creation
[   335][E][vfs_api.cpp:104] open(): /littlefs/ledmap2.json does not exist, no permits for creation
[   341][E][vfs_api.cpp:104] open(): /littlefs/ledmap3.json does not exist, no permits for creation
[   350][E][vfs_api.cpp:104] open(): /littlefs/ledmap4.json does not exist, no permits for creation
[   359][E][vfs_api.cpp:104] open(): /littlefs/ledmap5.json does not exist, no permits for creation
[   368][E][vfs_api.cpp:104] open(): /littlefs/ledmap6.json does not exist, no permits for creation
[   376][E][vfs_api.cpp:104] open(): /littlefs/ledmap7.json does not exist, no permits for creation
[   385][E][vfs_api.cpp:104] open(): /littlefs/ledmap8.json does not exist, no permits for creation
[   394][E][vfs_api.cpp:104] open(): /littlefs/ledmap9.json does not exist, no permits for creation
[   403][E][vfs_api.cpp:104] open(): /littlefs/ledmap10.json does not exist, no permits for creation
[   412][E][vfs_api.cpp:104] open(): /littlefs/ledmap11.json does not exist, no permits for creation
[   420][E][vfs_api.cpp:104] open(): /littlefs/ledmap12.json does not exist, no permits for creation
[   429][E][vfs_api.cpp:104] open(): /littlefs/ledmap13.json does not exist, no permits for creation
[   438][E][vfs_api.cpp:104] open(): /littlefs/ledmap14.json does not exist, no permits for creation
[   447][E][vfs_api.cpp:104] open(): /littlefs/ledmap15.json does not exist, no permits for creation
[   457][E][vfs_api.cpp:104] open(): /littlefs/palette0.json does not exist, no permits for creation
[   465][E][vfs_api.cpp:104] open(): /littlefs/ledmap.json does not exist, no permits for creation
Ada```

softhack007, 2023-03-05:
> Soo, I'm following this and so far I have this output log... any idea?

Yes `/littlefs/ledmap*.json does not exist` is expected behaviour. No problem.

WAPEETY, 2023-03-05:
> > Soo, I'm following this and so far I have this output log... any idea?
> 
> Yes `/littlefs/ledmap*.json does not exist` is expected behaviour. No problem.

Yes but it'repeted every second I think the ESP is bootlooping

softhack007, 2023-03-05:
You mean the complete block including `load:0x403d0000,len:0x24e4 entry 0x403ce000` is repeating, or just the `open().... does not exists` part?

In both cases, you could build with `-D WLED_DEBUG` added to you build_flags, to get more information about what's happening.

WAPEETY, 2023-03-05:
> 

the whole block I posted is repeating, btw I'll try wled debug ASAP

WAPEETY, 2023-03-05:
```---WLED 0.14.0-b2 2302150 INIT---
esp32 v4.4.2
arduino-esp32 v2.0.5
CPU:   ESP32-C3 rev.3, 1 core(s), 160MHz.
FLASH: 4MB, Mode 3 (DOUT), speed 80MHz.
heap 234504
PIN ALLOC: Pin 21 successfully allocated by 0x89 (137)
Registering usermods ...
Reading config
Reading settings from /wsec.json...
JSON buffer locked. (3)
JSON buffer released. (3)
JSON buffer locked. (1)
Reading settings from /cfg.json...
Removing all.
PIN ALLOC: Pin 0 successfully allocated by 0x85 (133)
Starting usermod config.
JSON buffer released. (1)
Initializing strip
[  2848][E][vfs_api.cpp:104] open(): /littlefs/ledmap1.json does not exist, no permits for creation
[  2855][E][vfs_api.cpp:104] open(): /littlefs/ledmap2.json does not exist, no permits for creation
[  2864][E][vfs_api.cpp:104] open(): /littlefs/ledmap3.json does not exist, no permits for creation
[  2873][E][vfs_api.cpp:104] open(): /littlefs/ledmap4.json does not exist, no permits for creation
[  2881][E][vfs_api.cpp:104] open(): /littlefs/ledmap5.json does not exist, no permits for creation
[  2890][E][vfs_api.cpp:104] open(): /littlefs/ledmap6.json does not exist, no permits for creation
[  2899][E][vfs_api.cpp:104] open(): /littlefs/ledmap7.json does not exist, no permits for creation
[  2908][E][vfs_api.cpp:104] open(): /littlefs/ledmap8.json does not exist, no permits for creation
[  2916][E][vfs_api.cpp:104] open(): /littlefs/ledmap9.json does not exist, no permits for creation
[  2925][E][vfs_api.cpp:104] open(): /littlefs/ledmap10.json does not exist, no permits for creation
[  2934][E][vfs_api.cpp:104] open(): /littlefs/ledmap11.json does not exist, no permits for creation
[  2943][E][vfs_api.cpp:104] open(): /littlefs/ledmap12.json does not exist, no permits for creation
[  2952][E][vfs_api.cpp:104] open(): /littlefs/ledmap13.json does not exist, no permits for creation
[  2961][E][vfs_api.cpp:104] open(): /littlefs/ledmap14.json does not exist, no permits for creation
[  2969][E][vfs_api.cpp:104] open(): /littlefs/ledmap15.json does not exist, no permits for creation
No busses, init default
PIN ALLOC: Pin 2 successfully allocated by 0x82 (130)
Successfully inited strip 0 (len 30) with type 22 and pins 2,255 (itype 21)
Loading custom palettes
[  3000][E][vfs_api.cpp:104] open(): /littlefs/palette0.json does not exist, no permits for creation
Loading custom ledmaps
[  3013][E][vfs_api.cpp:104] open(): /littlefs/ledmap.json does not exist, no permits for creation
Usermods setup
IMPROV�
lastReconnectAttempt == 0
No connection configured.
Opening accessESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0xf (BROWNOUT_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DOUT, clock div:1
load:0x3fcd6100,len:0x38c
load:0x403ce000,len:0x6ac
load:0x403d0000,len:0x24e4
entry 0x403ce000
```

lsiudut, 2023-03-05:
According to the reset reason - BROWNOUT_RST - you're not providing enough power to your module, voltage drops too much and it eventually resets. Lookup what brownout detector is and how it works.

WAPEETY, 2023-03-05:
> 

Yess, I was using programmer 5v :)
Seems to work, Thanks!

WAPEETY, 2023-03-05:
Ok so, 
- for some reason instead of using GPIO 2 (described by the board manufacturer) i'm using GPIO 10 (Idk if is a bug or not)

but it seems to work... 
I'll investigate further on possible problems

for the info about the board this is the website: https://www.electrodragon.com/product/esp-led-strip-board/?attribute_pa_esp=esp32-c3-12-2mb

here are schematics: https://w.electrodragon.com/w/ESP_LED_Board_HDK

metawops, 2023-04-25:
Still no official support for ESP32-C3 boards? (I got the cool tiny USB-C one from Seeed and got the "Your ESP32-C3 board is not supported" message when trying to flash. 😞)

AllExHH, 2023-04-25:
I have 3 of them running, but i have to check, top find my ini. Since i
have Reinstall my complete pc

blazoncek, 2023-04-25:
Dear people, please read all posts above.
C3, S2 and S3 are supported, though still in experimental phase.
You can compile sources yourself or use binaries provided by @srg74 or @wled-install

metawops, 2023-04-25:
Thanks for the hint, @blazoncek – unfortunately, I still get the "... C3 board not supported" error from the https://wled-install.github.io/ page. 😞 
So I assume there's currently no installer web page available that automagically does everything for us, correct?
Could you then please point me to a little tutorial on how to install a binary manually onto my Seeed ESP32-C3 board? (I'm on a Mac, btw.) That would be very kind! 😊 
Thank you! 🙏