wled/WLED #1942: ESP-01S settings are not saved

xfactory9025, 2021-05-03:
Hello there. I use esp-01s with 1mb memory. Wled downloaded the latest version. After entering the wifi settings, it works, but when you start it again, the settings are not saved, it requests a wifi password again.When you run the same file with nodemcu, the settings are saved. How can I make it work right with the Esp-01s?

Aircoookie, 2021-05-03:
Hi!
Please use the `WLED_0.12.0_ESP8266_1M.bin` binary, which does allow saving on ESP01 :)

xfactory9025, 2021-05-03:
The save problem has been fixed, but it often disconnects and takes a long time to reconnect. The effects do not work in the mobile browser and the mobile application, only color adjustment can be made. On the computer, all functions are running.

xfactory9025, 2021-05-04:
Effects work on ios 13 but not on ios 11.

sansillusion, 2021-05-04:
It seems ios is having issues with proper management of different z-depth
elements on web pages. Seems like an ios issue to me. They don't respect
web standards.

xfactory9025, 2021-05-06:
The connection drops after a certain period of time. Esp8266 going into sleep mode?

closed 2021-09-19 by stale[bot]