wled/WLED #1945: not connecting

h3cker10632, 2021-05-04:
I used wled for a while and i didnt have any problems to start off then after a while i couldnt connect to my lights from my computer but i could from my phone now its i cant connect from either i re-flashed the board 2 times already i use the node mcu 8266 and i dont know if its the board or the lastest update with wled.

either way this project is amazing and you guys did a great job on it i love it.
Thank you

IbRuthless, 2021-05-04:
Hi,

I'm having same issue, everything working fine for over a month, then I noticed the other day it wasn't showing up in home assistant, rebooted and it was fine. Today though same happened again, this time though couldn't connect with HA, Mobile or PC. So I reflashed the D1 Mini v1, now I can't login to the WLED-AP, reflashed with older version of WLED same issues.

Any ideas?

huggy-d1, 2021-05-05:
To the issue poster, if WLED is allowing wifi sleep mode, disable that.

To the d1-mini-reflash-failure poster, erase the d1-mini before reflashing.  New / old versions are not compatible.  Anything before v0.12 I mean by "old" versions.  If it has not had v0.12 installed, it should be more compatible.  For sure if it the d1-mini has only had v0.10 and below, reflashing _should_ not make a difference, but is still worth a shot.

joke2600, 2021-05-06:
I am having a similar issue.  I am running wled off of an 8266 nodemcu and for the first few months things were great! then I started noticing I needed to power cycle things once every other week now it seems that more often than not i cant connect to it via home assistant or via the webpage. i recently (today) upgraded from 11 to 12 and still same issue. Is there a debug version I can run that we can use to get some logs out of to try to figure out what this issue is all about?

huggy-d1, 2021-05-06:
I would suspect (guess) the 2.4GHz wifi network is getting busy, or the signal strength has dropped because the wifi AP or router is using a different channel that isn't getting through to the nodemcu as well.  Just a guess.

IbRuthless, 2021-05-06:
I'm still new to flashing these board so when you say erase the d1-mini, you couldn't explain how to do this or point me in the right direction.

joke2600, 2021-05-06:
is there anyway to check the signal strength ?

h3cker10632, 2021-05-06:
what about the node mcu esp 8266 is there any work around or fix to the overall connecting problem would it be the wifi maybe or bad boards, or even a software bug?

joke2600, 2021-05-07:
So questions. How do we go about getting a debug build so we can try to see some logs about what is happening?
and other questions for those experiencing this issue What kind of setup do you have?
I have 2
1st on a nodemcu 8266 and i have wifi sleep disabled and is connected to homeassistant and nodered inside homeassistant
2nd nodemcu 8266 again but is being used in conjuction with hyperion to control lights around the tv. this one is in homeassistant but not nodered and wifi sleep is enabled on this one. this one is just starting to become unavailable

h3cker10632, 2021-05-07:
I personally do not know anything about home servers and connecting devices to them or even using other softwares with them I used only the standard wled UI and I think from there you can connect other things with it if not take a look at the wiki and see if there are any pointer for you

joke2600, 2021-05-07:
You can connect toe home assistant and such I have that done I'm just trying to gather info so hopefully we can find whats common and figure out how to get the issue fixed.

IbRuthless, 2021-05-08:
I've managed to get mine up and running after a reflash 0.12v, however I'm back to teh same issue, WLED APP and moblie browser (chrome) wont connect. But I can connect with firefox on my windows 10 pc.

closed 2021-09-19 by stale[bot]