wled/WLED #1948: State value to set for "wled.0.483fda49c0fd._info.mac" has to be type "number" but received type "string"

OlfrygtACAB, 2021-05-06:
<html>
<body>

wled.0 | 2021-05-06 20:35:27.254 | info | (1832) State value to set for "wled.0.ecfabcc232d5.seg.0.col.2" has to be one of type "string", "number", "boolean" but received type "object"
-- | -- | -- | --
wled.0 | 2021-05-06 20:35:27.242 | info | (1832) State value to set for "wled.0.ecfabcc232d5.seg.0.col.1" has to be one of type "string", "number", "boolean" but received type "object"
wled.0 | 2021-05-06 20:35:27.234 | info | (1832) State value to set for "wled.0.ecfabcc232d5.seg.0.col.0" has to be one of type "string", "number", "boolean" but received type "object"
wled.0 | 2021-05-06 20:35:27.127 | info | (1832) State value to set for "wled.0.ecfabcc232d5._info.mac" has to be type "number" but received type "string"
wled.0 | 2021-05-06 20:35:27.051 | info | (1832) State value to set for "wled.0.ecfabcc232d5._info.leds.pin" has to be one of type "string", "number", "boolean" but received type "object"

</body>
</html>

huggy-d1, 2021-05-06:
If you have recently installed WLED, I recommend fully erasing the ESP, then loading the firmware again.

Aircoookie, 2021-05-07:
Hi! The received types of "object" (rather, an array in most cases) and "string" for the MAC address are the correct ones as per my API documentation. What software are you using for interacting with the JSON API that created these logs?

closed 2021-05-10 by Aircoookie