wled/WLED #1949: Platform io intermittently updating BIN

ThatBigPrint, 2021-05-06:
using the latest build and platform io install I have just started playing with my on complies and for whatever reason platform io isn't a producing/updating bin files and b when I delete the build output as a test to prove it's not updating it won't reproduce that file with a new bin file? but it works every now and then I have rebooted etc and no errors come up ?

ThatBigPrint, 2021-05-06:
one more thing this is on mac

huggy-d1, 2021-05-06:
The first time platform io compiles for a wled build, it probably fails while it works out library versions and dependencies, and will take a fairly long time.

The 2nd time and compiles are that, it should success, and should be pretty fast.

Is that at all helpful?

ThatBigPrint, 2021-05-06:
kinda, it just wont update/replace the bin file with the new version when I build 

to prove this I removed the build folder it produces on its first-ever build on a project and it never tried to replace it it just uploads the old data to the board directly too ?

ThatBigPrint, 2021-05-07:
so a bit of an update if i clone the hole project and open it it will spit out a bin etc

ThatBigPrint, 2021-05-07:
I still cant get the temperature working it's on pin4 and it works with the basic example sketch. I'm just not sure this is actually dumping a valid bin file

ThatBigPrint, 2021-05-07:
Can someone do me a favour and compile for an esp32dev with temp sensor on pin 4 and autosave working as a test, please? I think platformio is broken

Aircoookie, 2021-05-07:
Hi, I've occasionally experienced that issue too, it doesn't seem like a new binary is produced if no code changes are detected. You can try changing the `VERSION` in the top of `wled.h` by one digit and it should provide a valid bin to the output folder :)

ThatBigPrint, 2021-05-07:
That sounds a good plan, thank you very much

Also, have you had an issue with the temperature not working and showing
readings in the info window?

ThatBigPrint, 2021-05-07:
Up Rev worked ! however getting this temp sensor to work is not going well

ThatBigPrint, 2021-05-07:
`[platformio]
; default_envs = d1_mini
default_envs = esp32dev

[env:esp32dev]
[env:esp32dev_usermod_dallas_temperature_C]
board = esp32dev
platform = espressif32@3.2
build_unflags = ${common.build_unflags}
build_flags =
    ${common.build_flags_esp8266} -D USERMOD_DALLASTEMPERATURE -D USERMOD_DALLASTEMPERATURE_CELSIUS
    ${common.build_flags_esp32} -D USERMOD_AUTO_SAVE -D AUTOSAVE_PRESET_NUM=1

upload_speed = 460800
lib_ignore =
  ESPAsyncTCP
  ESPAsyncUDP

[env:d1_mini]
board = d1_mini
platform = ${common.platform_wled_default}
platform_packages = ${common.platform_packages}
upload_speed = 460800
board_build.ldscript = ${common.ldscript_4m1m}
build_unflags = ${common.build_unflags}
build_flags =
    ${common.build_flags_esp8266} 
    -D USERMOD_MODE_SORT
    -D USERMOD_ROTARY_ENCODER_UI -D ENCODER_DT_PIN=12 -D ENCODER_CLK_PIN=14 -D ENCODER_SW_PIN=13
    -D USERMOD_AUTO_SAVE -D AUTOSAVE_PRESET_NUM=1
    -D LEDPIN=3 -D BTNPIN=0
monitor_filters = esp8266_exception_decoder

[env]
lib_deps =
    fastled/FastLED @ 3.3.2
    NeoPixelBus @ 2.6.0
    ESPAsyncTCP @ 1.2.0
    ESPAsyncUDP
    AsyncTCP @ 1.0.3
    IRremoteESP8266 @ 2.7.3
    https://github.com/lorol/LITTLEFS.git
    https://github.com/Aircoookie/ESPAsyncWebServer.git @ ~2.0.0
    U8g2@~2.27.2
    Wire
    milesburton/DallasTemperature@^3.9.0
    OneWire@~2.3.5`

ThatBigPrint, 2021-05-07:
autosave usermod isn't even working I must be missing something simple up here

Aircoookie, 2021-05-17:
@ThatBigPrint sorry for the issues! Temperature mod should be fixed now, give compiling the latest master code a try and let me know how it goes :)

ThatBigPrint, 2021-05-17:
Ahh, so there was a known issue? I was pulling hair out!

blazoncek, 2021-05-17:
There is no more need to define usermod parameters during compile time.
Once you have flashed ESP opens Settings page and then Usermods. If the output is empty, just hit save and then re-open page.
After that you will be able to configure usermods (if the page is still empty no compatible usermods have been compiled in).

Aircoookie, 2021-05-17:
Depends on what you count as "known" :)
I experienced it, but did not think it was a software problem, attributed it to my sensor not working correctly.
I'm very glad that @blazoncek implemented it differently in #1951 and it works a lot better now!

ThatBigPrint, 2021-05-19:
I must still be missing something.... as I understand for temperature for instance 

you copy the example platformio_override.ini into the root? I have tried to uncomment options within that folder and it just errors out on comp if I leave them all commented it goes through fine please see below, I even manually changed the board type from d1 to the one I'm using to no avail... apologies for being slow with this I appreciate the patience 

; Options
; -------
USERMOD_DALLASTEMPERATURE                      - define this to have this user mod included wled00\usermods_list.cpp
; USERMOD_DALLASTEMPERATURE_CELSIUS              - define this to report temperatures in degrees celsius, otherwise fahrenheit will be reported
; USERMOD_DALLASTEMPERATURE_MEASUREMENT_INTERVAL - the number of milliseconds between measurements, defaults to 60 seconds
; USERMOD_DALLASTEMPERATURE_FIRST_MEASUREMENT_AT - the number of milliseconds after boot to take first measurement, defaults to 20 seconds
;
[env:esp32dev_usermod_dallas_temperature_C]
extends = env:esp32dev
build_flags = ${common.build_flags_esp32dev} -D USERMOD_DALLASTEMPERATURE -D USERMOD_DALLASTEMPERATURE_CELSIUS
lib_deps = ${env.lib_deps}
    milesburton/DallasTemperature@^3.9.0
    OneWire@~2.3.5

blazoncek, 2021-05-20:
> [env:esp32dev_usermod_dallas_temperature_C]
> extends = env:esp32dev
> build_flags = ${common.build_flags_esp32dev} -D USERMOD_DALLASTEMPERATURE
> lib_deps = ${env.lib_deps}
> OneWire@~2.3.5

You only need these in your platformio.ini to enable Dallas temperature sensors.
If you still struggle try one of precompiled experimental binaries from @srg74 or try to use #Bot on WLED Disord.

ThatBigPrint, 2021-05-20:
thanks for that will i need to change the esp32dev tag I swapped with d1 mini ?

blazoncek, 2021-05-20:
Critical things are -D USERMOD_DALLASTEMPERATURE in `build_flags` portion and OneWire@~2.3.5 in `lib_deps`.

blazoncek, 2021-05-21:
You must be doing something wrong.
@srg74 binaries are tested and work, there is no such thing as IO conflict causing boot loop except by bad wiring. If you still think this may be the case, please disconnect all wires from ESP and connect it to your computer using USB cable only then erase the flash memory on your ESP first, prior to uploading new firmware, then try flasing new firmware using esptool.

I got the impression that you are not familliar with PlatformIO so please learn the basics of VS Code and PlatformIO. I am sorry if I am mistaken.

Following is a complete section for ESP8266 (on Wemos D1 mini compatible dev board) in platformio.ini (or platformio_override.ini) that I am using and works as well on Windows and on mac.
You should just copy-paste it into your platformio.ini (or platformio_override.ini).

```
[env:d1_mini_temp]
extends = env:d1_mini
build_flags = ${common.build_flags_esp8266}
  -D USERMOD_DALLASTEMPERATURE
  -D TEMPERATURE_PIN=13 # (D7)
lib_deps = ${env.lib_deps}
    OneWire@~2.3.5
```

ThatBigPrint, 2021-05-21:
So it's working now kinda, but won't let me use pin 4 it just shows red but I don't think it's being used elsewhere In the files?

ThatBigPrint, 2021-05-21:
Got it sorted now it had a conflict with ir led

closed 2021-09-26 by stale[bot]