wled/WLED #1953: Brightness Preview for Segments is skaling to much

timae146, 2021-05-08:
**Describe the bug**
The Brightness Preview for Segments scales to much. If you set the seg. brigness value to low you cant see sth. in the WebGui Preview.

**To Reproduce**
See the gif

**Expected behavior**
Would be nice if the scaling could be modified same like the Preview of the Main brightness 

**WLED version**
 - Board: Wemos D1
 - Version 0.12.0
 - Format Binary

[image]

Aircoookie, 2021-05-10:
Hi!
Both sliders scale the output equally - if you set both to 50%, the resulting output will be 25%. This also means that if you put one slider on minimum, the scaling will cause the LEDs to appear off if the other slider is not at maximum. When changing the brightness of an individual segment, I would recommend leaving the master brightness slider on maximum.

There is indeed no use case for two sliders when running a single segment, but it is useful with multiple segments to set different brightness values for different parts of the installation :)

timae146, 2021-05-10:
hey. :) 

that's what I thought too. main slider to max and adjust the individual segments, but the disadvantage by this is that you cant the a preview if the brightness is to low.

[image]
main slider max (peek activated) you can´t see a preview with the segment slider value ~<20

this problem doesnt exist if you only adjust the main slider:
[image]

closed 2021-09-19 by stale[bot]